autowarefoundation/autoware #5174: Use same tag for multiple cpu architectures in docker images

xmfcx, 2024-09-05:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

[image]

Right now we have different tags for `arm64` and `amd64` images.

This is redundant since docker supports multiple cpu arch's in same tag.

Example:
[image]

### Purpose

Simplify the docker image tags by making use of modern docker features.

### Possible approaches

https://www.docker.com/blog/multi-arch-build-and-images-the-simple-way/

### Definition of done

Image tags don't have `arm64` or `amd64` keywords in them.

youtalk, 2024-09-05:
It is already performed by daily workflow https://github.com/autowarefoundation/autoware/blob/main/.github/workflows/update-docker-manifest.yaml. 

https://github.com/autowarefoundation/autoware/pkgs/container/autoware/268585026?tag=latest-runtime-cuda

xmfcx, 2024-09-05:
However, my question is: why are we generating architecture-specific tags initially and then merging them later? Wouldn't it be cleaner and more efficient to push a single tag directly, without separating them at the start? This could help in keeping the package repository more organized.

youtalk, 2024-09-05:
I don't know how to do it properly because the CI runners are different from machines and architectures so that I think the manifest could be also different.
I hope @oguzkaganozt knows more than me.

youtalk, 2024-09-05:
This is not my emergent issue. 
I'm waiting for your contribution not only by discussion but by PR. 
Thanks.

oguzkaganozt, 2024-09-06:
I will investigate try to find out more straight-forward solution to this.

oguzkaganozt, 2024-09-24:
This should be resolved with https://github.com/autowarefoundation/autoware/pull/5256

closed 2024-09-24 by oguzkaganozt