autowarefoundation/autoware #5175: Remove the latest keyword from docker tags

xmfcx, 2024-09-05:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

@youtalk -san, I think the latest keyword is also becoming redundant, what do you think of removing it?
Is there a benefit in keeping the latest keyword?

| Before | After |
| - | - |
| `autoware:latest-base` | `autoware:base` |
| `autoware:latest-core-devel` | `autoware:core-devel` |
| `autoware:latest-universe-devel` | `autoware:universe-devel` |
| `autoware:latest-devel` | `autoware:devel` |
| `autoware:latest-universe` | `autoware:universe` |

### Purpose

To not have redundant keywords in images.

### Possible approaches

-

### Definition of done

- [ ] Docker image tags don't have `latest` keyword anymore.
- [ ] Dependent repos are updated.

youtalk, 2024-09-05:
According to the discussion https://github.com/orgs/autowarefoundation/discussions/4995 I approved.
However I would like to remain the `autoware:latest` tag to be the same as the `autoware:universe` tag for a while.

xmfcx, 2024-09-05:
It's alright to have `autoware:latest` :+1:

oguzkaganozt, 2024-09-05:
`autoware:latest` can be good and a needful simplification to users

youtalk, 2024-09-05:
This is not my emergent issue. 
I'm waiting for your contribution not only by discussion but by PR. 
Thanks.

oguzkaganozt, 2024-09-09:
Fix PR https://github.com/autowarefoundation/autoware/pull/5201

closed 2024-09-11 by oguzkaganozt