autowarefoundation/autoware #5176: Update the docker images readme file to match the new updates

xmfcx, 2024-09-05:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

https://github.com/autowarefoundation/autoware/tree/main/docker#readme is out of date after some new PRs merged or will be merged:
- https://github.com/autowarefoundation/autoware/pull/5164
- https://github.com/autowarefoundation/autoware/pull/5170
- https://github.com/autowarefoundation/autoware/pull/5171
- ...

### Purpose

To keep the readme up-to-date.

### Possible approaches

.

### Definition of done

Readme is updated.

youtalk, 2024-09-05:
I would like to update the documentation when the multi-containerization is fully migrated. It's within this month I hope.

youtalk, 2024-09-05:
Of course I really appreciate if someone helps the documentation. I think everyone who knows Docker can contribute.

xmfcx, 2024-09-06:
Closing since:
- https://github.com/autowarefoundation/autoware/pull/5185
is merged.