autowarefoundation/autoware #5182: Documentation how to use multi-containerized Autoware

youtalk, 2024-09-06:
Write about how to use Autoware's multi-container in the Autoware Documentation or README.md.

oguzkaganozt, 2024-09-06:
@youtalk -san if you can label `containers` and `openadkit` related tasks with these two labels it would be easy to find and discuss them within the WG meetings. Thanks !

oguzkaganozt, 2024-09-19:
1. **base**:
   - **Purpose**: Serves as the foundation for all other images.
   - **Includes**: Basic system setup, ROS installation, and essential tools.

2. **rosdep-depend**:
   - **Purpose**: Generates package dependency lists for core and common components.
   - **Includes**: rosdep tool and core/common source files.

3. **rosdep-universe-*-depend**:
   - **Purpose**: Generates package dependency lists for specific Autoware components (sensing, perception, localization, mapping, planning, control, vehicle, system).
   - **Includes**: Source files for respective components.

4. **core-devel**:
   - **Purpose**: Development environment for core Autoware components.
   - **Includes**: Core dependencies, build tools, and compiled core packages.

5. **universe-common-devel**:
   - **Purpose**: Development environment for common Autoware Universe components.
   - **Includes**: Common dependencies and compiled common packages.

6. **universe-*-devel**:
   - **Purpose**: Development environments for specific Autoware Universe components.
   - **Includes**: Dependencies and compiled packages for respective components.

7. **universe-devel**:
   - **Purpose**: Complete development environment for Autoware Universe.
   - **Includes**: All Autoware Universe components and their dependencies.

8. **universe-***:
   - **Purpose**: Runtime images for specific Autoware Universe components.
   - **Includes**: Runtime dependencies and compiled packages for respective components.

9. **universe**:
   - **Purpose**: Complete runtime image for Autoware Universe.
   - **Includes**: All runtime dependencies and compiled Autoware Universe packages.

oguzkaganozt, 2024-09-23:
Here are the two flavors of mermaid  diagrams that I am suggesting to update the documentation with: @youtalk @mitsudome-r @xmfcx @kaspermeck-arm 

**1-** 
```mermaid
flowchart TD
    %% Base Image
    A[Base]

    %% Development Stages
    subgraph Development[" "]
        B[Core Devel]
        C[Universe-Common Devel]
        subgraph Components["**Development Images**"]
            direction LR
            D[Sensing-Perception Devel]
            E[Localization-Mapping Devel ]
            F[Planning-Control Devel]
            G[Vehicle-System Devel]
        end
        H[Universe-Devel]
    end

    %% Final Images
    subgraph Final["**Runtime Images**"]
        direction LR
        I[Sensing-Perception]
        J[Localization-Mapping]
        K[Planning-Control]
        L[Vehicle-System]
        M[Universe]
    end

    %% Relationships
    A -->|FROM| B
    B -->|FROM| C
    C -->|FROM| Components
    C -->|FROM| H
    A -->|FROM| Final

    %% Component to Universe and Final Output relationships
    Components -.->|COPY| H
    Components -.->|COPY| Final

    %% Styling
    classDef base fill:#f39c12,stroke:#333,stroke-width:3px;
    classDef dev fill:#85c1e9,stroke:#1f618d,stroke-width:3px;
    classDef modular fill:#58d68d,stroke:#229954,stroke-width:3px,stroke-dasharray: 5 5;
    classDef monolithic fill:#58d68d,stroke:#229954,stroke-width:3px;

    class A base;
    class B,C,D,E,F,G,H dev;
    class I,J,K,L modular;
    class M monolithic;

    linkStyle default stroke:#34495e,stroke-width:2px,stroke-opacity:0.7;
```

**2-**

```mermaid
flowchart TD
    %% Relationships
    Base -->|FROM| Rosdep
    Base -->|FROM| Dev
    Base -->|FROM| Runtime

    %% Rosdep Details
    subgraph Rosdep
        RD[Rosdep-Depend]
        RU[Rosdep-Universe-Depend]
        subgraph RC[Component-Specific Rosdep]
            direction LR
            RSP[Sensing-Perception]
            RLM[Localization-Mapping]
            RPC[Planning-Control]
            RVS[Vehicle-System]
        end
    end

    %% Development Details
    subgraph Dev
        CD[Core-Devel]
        UCD[Universe-Common-Devel]
        subgraph CompDev[Component-Specific Devel]
            direction LR
            DSP[Sensing-Perception Devel]
            DLM[Localization-Mapping Devel]
            DPC[Planning-Control Devel]
            DVS[Vehicle-System Devel]
        end
        UD[Universe-Devel]
    end

    %% Runtime Details
    subgraph Runtime
        SP[Sensing-Perception]
        LM[Localization-Mapping]
        PC[Planning-Control]
        VS[Vehicle-System]
        U[Universe]
    end

    %% Additional Relationships
    RD -.->|COPY| CD
    RU -.->|COPY| UCD
    RC -.->|COPY| CompDev
    CD -.->|FROM| UCD
    UCD -->|FROM| CompDev
    CompDev -->|FROM| UD
    DSP -.->|COPY| SP
    DLM -.->|COPY| LM
    DPC -.->|COPY| PC
    DVS -.->|COPY| VS
    UD -.->|COPY| U

    %% Styling
    classDef default fill:#f9f9f9,stroke:#333,stroke-width:2px;
    classDef main fill:#f0f0f0,stroke:#333,stroke-width:3px;
    classDef base fill:#f39c12,stroke:#333,stroke-width:3px;
    classDef rosdep fill:#f1948a,stroke:#c0392b,stroke-width:2px;
    classDef dev fill:#85c1e9,stroke:#1f618d,stroke-width:2px;
    classDef runtime fill:#58d68d,stroke:#229954,stroke-width:2px;

    class Base base;
    class Rosdep,Dev,Runtime main;
    class RD,RU,RSP,RLM,RPC,RVS rosdep;
    class CD,UCD,DSP,DLM,DPC,DVS,UD dev;
    class SP,LM,PC,VS,U runtime;

    linkStyle default stroke:#666,stroke-width:2px;
```

kaspermeck-arm, 2024-09-23:
@oguzkaganozt 

The first option is a lot easier to comprehend and if you make a statement saying that the flow is simplified and doesn't contain `Rosdep-Depend` dependencies, that should be fine. Thanks!

oguzkaganozt, 2024-09-24:
> @oguzkaganozt
> 
> The first option is a lot easier to comprehend and if you make a statement saying that the flow is simplified and doesn't contain `Rosdep-Depend` dependencies, that should be fine. Thanks!

Thanks for the feedback !