autowarefoundation/autoware #5193: Docker Build `--devel-only` option is failing.

isouf, 2024-09-07:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

When running `./docker/build.sh --devel-only` it gives the following error:

`ERROR: could not find any target matching 'devel'`

### Expected behavior

Docker builds process starts without errors.

### Actual behavior

Docker build process fails.

`ERROR: could not find any target matching 'devel'`

### Steps to reproduce

Build Docker images using `./docker/build.sh --devel-only`.

### Versions

- OS: Ubuntu 22.04
- Autoware: Latest `main`

### Possible causes

The `./docker/build.sh` file was not updated as part of this commit: https://github.com/autowarefoundation/autoware/commit/7148b7769cc26c60410bba2aa49d18d058e886c4

The following line of the `./docker/build.sh` will have to be updated:
https://github.com/autowarefoundation/autoware/blob/main/docker/build.sh#L62

oguzkaganozt, 2024-09-09:
Hi @isouf thanks for reporting !

oguzkaganozt, 2024-09-09:
Hopefully [this ](https://github.com/autowarefoundation/autoware/pull/5199)will resolve it

closed 2024-09-11 by oguzkaganozt