autowarefoundation/autoware #5265: Obstacle_Stop_planner don't working

RezaHosseini80, 2024-09-20:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

when using obstacle stop planner instead of obstacle cruise planner it can not change operation mode and show this error :

[service_log_checker_node-5] [ERROR] [1726819007.429075993] [system.service_log_checker]: /api/operation_mode/change_to_autonomous: status code 1 'The target mode is not available. Please check the diagnostics.' (/default_ad_api/node/operation_mode)

and these logs are repeated continuously:

[component_container_mt-38] [INFO] [1726819066.577174532] [control.autonomous_emergency_braking]: [AEB] waiting for control predicted trajectory
[topic_state_monitor_node-13] [INFO] [1726819067.115392865] [system.topic_state_monitor_control_command_control_cmd]: /control/command/control_cmd has not received. Set ERROR in diagnostics.
[topic_state_monitor_node-12] [INFO] [1726819067.381593641] [system.topic_state_monitor_trajectory_follower_control_cmd]: /control/trajectory_follower/control_cmd has not received. Set ERROR in diagnostics.
[topic_state_monitor_node-11] [INFO] [1726819067.576380438] [system.topic_state_monitor_scenario_planning_trajectory]: /planning/scenario_planning/trajectory has not received. Set ERROR in diagnostics.
[topic_state_monitor_node-13] [INFO] [1726819070.115494059] [system.topic_state_monitor_control_command_control_cmd]: /control/command/control_cmd has not received. Set ERROR in diagnostics.
[component_container_mt-38] [INFO] [1726819070.266317108] [control.trajectory_follower.lane_departure_checker_node]: waiting for reference_trajectory msg...
[topic_state_monitor_node-12] [INFO] [1726819070.481496364] [system.topic_state_monitor_trajectory_follower_control_cmd]: /control/trajectory_follower/control_cmd has not received. Set ERROR in diagnostics.
[component_container_mt-38] [INFO] [1726819070.675251791] [control.trajectory_follower.controller_node_exe]: Waiting for trajectory data
[component_container_mt-38] [INFO] [1726819070.675310232] [control.trajectory_follower.controller_node_exe]: Control is skipped since input data is not ready.
[topic_state_monitor_node-11] [INFO] [1726819070.676377590] [system.topic_state_monitor_scenario_planning_trajectory]: /planning/scenario_planning/trajectory has not received. Set ERROR in diagnostics.

### Expected behavior

the Auto mode would be available in obstacle_stop_planner mode.

### Actual behavior

the Auto mode is  not available in obstacle_stop_planner mode.

### Steps to reproduce

1. set the motion_stop_planner_type to obstacle_stop_planner.
2. run the autoware.
3. set the initial and goal positions.
4. THE Auto mode is grayed out.

maxime-clem, 2024-09-20:
Which version of Autoware are you using ?
I could not reproduce using a recent version from today.
- `autoware.universe`: `59838b6e7552ef1bc61867f9dc65de4dfa472529`
- `autoware_launch`: `3a08555776df12eadd0e91867d338c4270932a44`

RezaHosseini80, 2024-09-22:
> Which version of Autoware are you using ? I could not reproduce using a recent version from today.
> 
> * `autoware.universe`: `59838b6e7552ef1bc61867f9dc65de4dfa472529`
> * `autoware_launch`: `3a08555776df12eadd0e91867d338c4270932a44`

i'm using the latest version

maxime-clem, 2024-09-23:
I tried on another machine after updating all repositories (`vcs pull src`) and rebuilding (`colcon build`) and I still cannot reproduce the issue.
I did have to do a clean build of the `autoware_signal_processing` package (`rm -rf install/autoware_signal_processing build/autoware_signal_processing` before rebuilding) to avoid issue.

I tried both changing `default_present.yaml`.
```diff
diff --git a/autoware_launch/config/planning/preset/default_preset.yaml b/autoware_launch/config/planning/preset/default_preset.yaml
index 5a3139e4c..2ec0d7ad2 100644
--- a/autoware_launch/config/planning/preset/default_preset.yaml
+++ b/autoware_launch/config/planning/preset/default_preset.yaml
@@ -105,7 +105,7 @@ launch:
 
   - arg:
       name: motion_stop_planner_type
-      default: obstacle_cruise_planner
+      default: obstacle_stop_planner
```

And also setting the parameter at launch:
```
ros2 launch autoware_launch planning_simulator.launch.xml [...] motion_stop_planner_type:=obstacle_stop_planner
```

If you still have the issue after a clean build, please share a full log of the launch (found in `$HOME/.ros/log/{DATE-USER-...}/launch.log`.

maxime-clem, 2024-09-23:
Thank you for sharing the log file. I can see the following issue:
```
1727095357.3660553 [ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'obstacle_stop_planner' of type 'motion_planning::ObstacleStopPlannerNode' in container '/planning/scenario_planning/lane_driving/motion_planning/motion_planning_container': Could not find requested resource in ament index
```
The name of the node (`obstacle_stop_planner`) and the type (`motion_planning::ObstacleStopPlannerNode`) do not match the current launch file: https://github.com/autowarefoundation/autoware.universe/blob/main/launch/tier4_planning_launch/launch/scenario_planning/lane_driving/motion_planning/motion_planning.launch.xml#L182

So I think you are missing this commit: https://github.com/autowarefoundation/autoware.universe/commit/a6f218a050f004bd33d90ec72b582da50eafa698
Please update and rebuild your workspace and the issue should be fixed.

RezaHosseini80, 2024-09-26:
> Thank you for sharing the log file. I can see the following issue:
> 
> ```
> 1727095357.3660553 [ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'obstacle_stop_planner' of type 'motion_planning::ObstacleStopPlannerNode' in container '/planning/scenario_planning/lane_driving/motion_planning/motion_planning_container': Could not find requested resource in ament index
> ```
> 
> The name of the node (`obstacle_stop_planner`) and the type (`motion_planning::ObstacleStopPlannerNode`) do not match the current launch file: https://github.com/autowarefoundation/autoware.universe/blob/main/launch/tier4_planning_launch/launch/scenario_planning/lane_driving/motion_planning/motion_planning.launch.xml#L182
> 
> So I think you are missing this commit: [autowarefoundation/autoware.universe@a6f218a](https://github.com/autowarefoundation/autoware.universe/commit/a6f218a050f004bd33d90ec72b582da50eafa698) Please update and rebuild your workspace and the issue should be fixed.

Thank you so much, it worked perfectly!