autowarefoundation/autoware #5289: Unable to pull docker image in Ubuntu 24.04

theksg, 2024-09-30:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I am using ROS 2 Jazzy and Ubuntu (wsl) 24.04.

Upon writing the command "docker pull ghcr.io/autowarefoundation/autoware:universe-devel-20240921", I get the below message
"no matching manifest for linux/amd64 in the manifest list entries"

### Expected behavior

Docker image should pull down

### Actual behavior

Getting message "no matching manifest for linux/amd64 in the manifest list entries"

### Steps to reproduce

1. Type command "docker pull ghcr.io/autowarefoundation/autoware:universe-devel-20240921"

### Versions

OS: Ubuntu 24.04
ROS 2 Jazzy

mitsudome-r, 2024-10-29:
@theksg Sorry for the confusion, but we have been renaming our docker images recently. 
If you want to use autoware from September 21, then try pulling: `docker pull ghcr.io/autowarefoundation/autoware:universe-20240921-amd64`.
If you want to pull the latest one, run `docker pull ghcr.io/autowarefoundation/autoware:universe-devel`

tranvansang, 2024-11-21:
@mitsudome-r 

Thanks for the comment.

I checked the latest tag [`universe-devel`](https://github.com/autowarefoundation/autoware/pkgs/container/autoware/309716459?tag=universe-devel) and [`universe-devel-cuda`](https://github.com/autowarefoundation/autoware/pkgs/container/autoware/309803340?tag=universe-devel-cuda).

The former supports `linux/amd64` and `linux/arm64`.
The latter supports only `linux/arm64`.

Is cuda build only available for arm64?

[image]

[image]