autowarefoundation/autoware #5355: `No space left on device` on `docker-build-and-push-arm64` workflow

youtalk, 2024-10-21:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Recently, the `docker-build-and-push-arm64` workflow has been repeatedly failing due to disk space issues.

https://github.com/autowarefoundation/autoware/actions/runs/11432044778/job/31801892052#step:8:24079
```
No space left on device
```

The free space of the arm64 self-hosted runner is 41GB, but since running the `jlumbroso/free-disk-space` action does not free up much space, it remains 12GB less than the free space of the amd64 GitHub-hosted runner (53GB) after executing the `jlumbroso/free-disk-space` action.

## amd64 platform (GitHub hosted runner)
- Before running `free-disk-space` action
```shell
$ dh -h /

Filesystem      Size  Used Avail Use% Mounted on
/dev/root        73G   52G   22G  71% /
```

- After running `free-disk-space` action
```shell
$ dh -h /

Filesystem      Size  Used Avail Use% Mounted on
/dev/root        73G   21G   53G  29% /
```

## aarch64 platform (Self hosted runner using AWS)
- Before running `free-disk-space` action
```shell
$ dh -h /

Filesystem      Size  Used Avail Use% Mounted on
/dev/root        97G   56G   41G  58% /
```

- After running `free-disk-space` action
```shell
$ dh -h /

Filesystem      Size  Used Avail Use% Mounted on
/dev/root        97G   56G   41G  58% /
```

### Expected behavior

`docker-build-and-push-arm64` jobs pass.

### Actual behavior

`docker-build-and-push-arm64` jobs failed.

### Steps to reproduce

1. Run `docker-build-and-push-arm64`

### Additional context

Is it possible to increase the disk size of the arm64 self-hosted runner? If we can increase it by 20GB, this issue can be resolved.

xmfcx, 2024-10-21:
@youtalk -san I will quickly investigate and report my findings.

xmfcx, 2024-10-21:
## Large files

### Docker volumes

```
ncdu 1.14.1 ~ Use the arrow keys to navigate, press ? for help
--- /var/lib/docker/volumes ------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
                         /..
    9.0 GiB [##########] /buildx_buildkit_builder-7d632cfb-a06d-4259-80c5-247e52d260e80_state
    8.3 GiB [######### ] /buildx_buildkit_builder-c93faaab-1668-4f10-8000-837c665b027c1_state
    7.8 GiB [########  ] /buildx_buildkit_remote-builder-arm64_state
    7.6 GiB [########  ] /buildx_buildkit_builder-34e5ddee-c91a-48b4-b273-618b8b91b6b01_state
```

For now, I've cleaned up all docker volumes/images/containers.

I'm not sure if these were left-over from old times, or recent.

I can create a cleanup script for these, if they appear and accumulate again.

Related: https://docs.github.com/en/actions/hosting-your-own-runners/managing-self-hosted-runners/running-scripts-before-or-after-a-job#about-pre--and-post-job-scripts

### Runner working directory

```
--- /home/ubuntu/actions-runner/_work --------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
                         /..
    7.0 GiB [##########] /autoware.universe
    2.5 GiB [###       ] /pix-openadkit
    2.1 GiB [##        ] /autoware_tools
    1.4 GiB [#         ] /autoware
    1.4 GiB [#         ] /openadkit_demo.autoware
```

To clean up these, we could utilize:

https://github.com/marketplace/actions/clean-self-hosted-runner

## Final action

Nevertheless, I will increase the size of the drive space from 100GB to 200GB.

xmfcx, 2024-10-21:
@youtalk -san, I've followed https://docs.aws.amazon.com/ebs/latest/userguide/recognize-expanded-volume-linux.html and I think it is done.

The disk space is now `194GB`.

```
ubuntu@ip-192-168-22-5:~$ df -h
Filesystem       Size  Used Avail Use% Mounted on
/dev/root        194G   25G  170G  13% /
```

Once an action like this:
- https://github.com/autowarefoundation/autoware/actions/runs/11432044778/job/31801892052#step:8:24079
finishes successfully, we can confirm and close this issue.

xmfcx, 2024-10-21:
Old issue link for reference:
- https://github.com/autowarefoundation/autoware/issues/2412

youtalk, 2024-10-21:
@xmfcx The no space error still occurred though...

xmfcx, 2024-10-21:
@youtalk -san, where did it occur?

youtalk, 2024-10-21:
@xmfcx Oh I'm sorry but it is no problem them! Thank you for your rapid reaction.

youtalk, 2024-10-21:
Once https://github.com/autowarefoundation/autoware/actions/runs/11437207211 will be succeeded, I will close this issue.

youtalk, 2024-10-21:
The heavier `cuda` job was OK. https://github.com/autowarefoundation/autoware/actions/runs/11437207211/job/31816255974
So we can close this. Thank you!