autowarefoundation/autoware #5523: stderr: autoware_tensorrt_common

shuofeng1, 2024-12-02:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I'm trying to build the main branch and I found an error. If you don't use deb to install cuda, cudnn and tensorrt, stderr: autoware_tensorrt_common will appear.
CMake Warning at CMakeLists.txt:17 (message):
  cuda, cudnn, tensorrt libraries are not found
My actual output found tensorrt not found. I have configured export LD_LIBRARY_PATH=/opt/TensorRT-8.6.1.6/lib:$LD_LIBRARY_PATH in the .bashrc file.
OS: Ubuntu 22.04.5 LTS
Processor: Intel® Core™ i7-10700F CPU @ 2.90GHz × 16
GPU: GeForce RTX 3060
OS Type: 64-bit
cuda version 12.0, cudnn8.9.7, tensorrt8.6.1.6
how can i solve

### Expected behavior

colcon build success

### Actual behavior

Summary: 347 packages finished [52min 36s]
  29 packages had stderr output: autoware_costmap_generator autoware_elevation_map_loader autoware_freespace_planning_algorithms autoware_image_projection_based_fusion autoware_lanelet2_extension_python autoware_lidar_apollo_instance_segmentation autoware_lidar_centerpoint autoware_lidar_transfusion autoware_overlay_rviz_plugin autoware_radar_scan_to_pointcloud2 autoware_shape_estimation autoware_smart_mpc_trajectory_follower autoware_surround_obstacle_checker autoware_tensorrt_classifier autoware_tensorrt_common autoware_tensorrt_yolox bag_time_manager_rviz_plugin eagleye_can_velocity_converter eagleye_coordinate eagleye_geo_pose_fusion eagleye_rt lanelet2_extension_python nebula_ros pointcloud_to_laserscan system_monitor yabloc_common yabloc_image_processing yabloc_particle_filter yabloc_pose_initializer

### Steps to reproduce

colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release

amadeuszsz, 2024-12-03:
Hi @shuofeng1 
Did you use [setup-dev-env.sh](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/source-installation/#how-to-set-up-a-development-environment) script for installing dependencies? Nevertheless, we suggest to use CUDA [12.3](https://github.com/autowarefoundation/autoware/blob/main/amd64.env#L6). If you would like to check which library is missing, the fastest way will be modifying `CMakeLists.txt` and check which of `CUDAToolkit_FOUND`,  `CUDNN_FOUND`, `TENSORRT_FOUND` fails. 

It's hard to localize source of issue in customized environment, therefore I can recommend you using Docker [containers](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/docker-installation/).

shuofeng1, 2024-12-03:
I'm not using the setup-dev-env.sh script 
except for the version above
I've also tried multiple versions
cuda12.3, cudnn8.9, tensorrt8.6;
cuda11.8, cudnn8.8, tensorrt8.6.
I tried outputting CUDAToolkit_FOUND, CUDNN_FOUND, TENSORRT_FOUND. In every version, TENSORRT_FOUND is false. I'm just a little confused if I need any other behavior after setting LD_LIBRARY_PATH=/opt/TensorRT-8.6.1.6/lib:$LD_LIBRARY_PATH using the tar installation method?
I am using the main branch

shuofeng1, 2024-12-03:
> Hi  你好@shuofeng1 Did you use [setup-dev-env.sh](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/source-installation/#how-to-set-up-a-development-environment) script for installing dependencies? Nevertheless, we suggest to use CUDA [12.3](https://github.com/autowarefoundation/autoware/blob/main/amd64.env#L6). If you would like to check which library is missing, the fastest way will be modifying `CMakeLists.txt` and check which of `CUDAToolkit_FOUND`, `CUDNN_FOUND`, `TENSORRT_FOUND` fails.您是否使用 [setup-dev-env.sh](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/source-installation/#how-to-set-up-a-development-environment) 脚本来安装依赖项？尽管如此，我们建议使用 CUDA [12.3](https://github.com/autowarefoundation/autoware/blob/main/amd64.env#L6)。如果您想检查缺少哪个库，最快的方法是修改`CMakeLists.txt`并检查`CUDAToolkit_FOUND`中的哪个库`CUDNN_FOUND` `TENSORRT_FOUND`失败。
> 
> It's hard to localize source of issue in customized environment, therefore I can recommend you using Docker [containers](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/docker-installation/).在自定义环境中很难定位问题根源，因此我建议您使用 Docker [容器](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/docker-installation/)。

I'm not using the setup-dev-env.sh script
except for the version above
I've also tried multiple versions
cuda12.3, cudnn8.9, tensorrt8.6;
cuda11.8, cudnn8.8, tensorrt8.6.
I tried outputting CUDAToolkit_FOUND, CUDNN_FOUND, TENSORRT_FOUND. In every version, TENSORRT_FOUND is false. I'm just a little confused if I need any other behavior after setting LD_LIBRARY_PATH=/opt/TensorRT-8.6.1.6/lib:$LD_LIBRARY_PATH using the tar installation method?
I am using the main branch

amadeuszsz, 2024-12-03:
> I'm not using the setup-dev-env.sh script except for the version above I've also tried multiple versions cuda12.3, cudnn8.9, tensorrt8.6; cuda11.8, cudnn8.8, tensorrt8.6. I tried outputting [...]

Could you show us an output of `dpkg -l | grep nvinfer`? Expected output:
```
hi  libnvinfer-dev                                    8.6.1.6-1+cuda12.0                      amd64        TensorRT development libraries
hi  libnvinfer-headers-dev                            8.6.1.6-1+cuda12.0                      amd64        TensorRT development headers
hi  libnvinfer-headers-plugin-dev                     8.6.1.6-1+cuda12.0                      amd64        TensorRT plugin headers
hi  libnvinfer-plugin-dev                             8.6.1.6-1+cuda12.0                      amd64        TensorRT plugin libraries
hi  libnvinfer-plugin8                                8.6.1.6-1+cuda12.0                      amd64        TensorRT plugin libraries
hi  libnvinfer8                                       8.6.1.6-1+cuda12.0                      amd64        TensorRT runtime libraries
```

If problem after installing missing dependencies still exists, please try with Docker.

shuofeng1, 2024-12-03:
> > I'm not using the setup-dev-env.sh script except for the version above I've also tried multiple versions cuda12.3, cudnn8.9, tensorrt8.6; cuda11.8, cudnn8.8, tensorrt8.6. I tried outputting CUDAToolkit_FOUND, CUDNN_FOUND, TENSORRT_FOUND. In every version, TENSORRT_FOUND is false. I'm just a little confused if I need any other behavior after setting LD_LIBRARY_PATH=/opt/TensorRT-8.6.1.6/lib:$LD_LIBRARY_PATH using the tar installation method? I am using the main branch除了上面的版本外，我没有使用 setup-dev-env.sh 脚本，我还尝试了多个版本 cuda12.3、cudnn8.9、tensorrt8.6;CUDA11.8、CUDNN8.8、TensorRT8.6 的。我尝试输出 CUDAToolkit_FOUND、CUDNN_FOUND TENSORRT_FOUND。在每个版本中，TENSORRT_FOUND 都是 false。我只是有点困惑，在使用 tar 安装方法设置 LD_LIBRARY_PATH=/opt/TensorRT-8.6.1.6/lib：$LD_LIBRARY_PATH 后是否需要任何其他行为？我正在使用 main 分支
> 
> Could you show us an output of `dpkg -l | grep nvinfer`? Expected output:您能给我们看一下 `dpkg -l | grep nvinfer` 的输出吗？预期输出：
> 
> ```
> hi  libnvinfer-dev                                    8.6.1.6-1+cuda12.0                      amd64        TensorRT development libraries
> hi  libnvinfer-headers-dev                            8.6.1.6-1+cuda12.0                      amd64        TensorRT development headers
> hi  libnvinfer-headers-plugin-dev                     8.6.1.6-1+cuda12.0                      amd64        TensorRT plugin headers
> hi  libnvinfer-plugin-dev                             8.6.1.6-1+cuda12.0                      amd64        TensorRT plugin libraries
> hi  libnvinfer-plugin8                                8.6.1.6-1+cuda12.0                      amd64        TensorRT plugin libraries
> hi  libnvinfer8                                       8.6.1.6-1+cuda12.0                      amd64        TensorRT runtime libraries
> ```
> 
> If problem after installing missing dependencies still exists, please try with Docker.如果安装后缺少的依赖项仍然存在问题，请尝试使用 Docker。

I installed tensorrt using a tar file. dpkg did not apply to me. Installing according to deb will complete the expected work. Thank you for your answer. If one day I successfully compile according to the tar installation method, I will put the solution here.

closed 2024-12-03 by amadeuszsz

vollkornholztuer, 2024-12-03:
I have the same issue. What i've noticed, is that when building from scratch (remove the folders: build/ log/ install/ before building), i get the same stderr's. When the Building is done, and i run same build command again, everything is built without errors/warnings.

@amadeuszsz, is that intended or not? Also, did the packages found the nvidia-dependencies in the second build-run or how does it exactly work?

Other than that, I'm also interested in a fix for that.

shuofeng1, 2024-12-04:
If you are installing using a tar file, please additionally add 
export TENSORRT_ROOT=/opt/TensorRT-8.6.1.6
and replace it with your actual installation path