autowarefoundation/autoware #5560: NvInferVersion.h - TensorRT-variables not found because of name change

vollkornholztuer, 2024-12-12:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

The package autoware_tensorrt_common doesn't build, because it expects old #define-variables from NvInferVersion.h. That also may apply to other packages dependent on TensorRT.

### Expected behavior

The package builds without any error.

### Actual behavior

The package autoware_tensorrt_common doesn't build and returns these errors to the terminal:

```
--- stderr: autoware_tensorrt_common                                                                                                                                                                    
In file included from /home/dekor-x/autoware/src/universe/autoware.universe/perception/autoware_tensorrt_common/src/tensorrt_common.cpp:15:
/home/dekor-x/autoware/src/universe/autoware.universe/perception/autoware_tensorrt_common/include/autoware/tensorrt_common/tensorrt_common.hpp:118:28: error: operator '>=' has no left operand
  118 | #if TENSORRT_VERSION_MAJOR >= 8
      |                            ^~
/home/dekor-x/autoware/src/universe/autoware.universe/perception/autoware_tensorrt_common/src/tensorrt_common.cpp:509:28: error: operator '>=' has no left operand
  509 | #if TENSORRT_VERSION_MAJOR >= 8
      |                            ^~
/home/dekor-x/autoware/src/universe/autoware.universe/perception/autoware_tensorrt_common/src/tensorrt_common.cpp:528:28: error: operator '<' has no left operand
  528 | #if TENSORRT_VERSION_MAJOR < 8
      |                            ^
/home/dekor-x/autoware/src/universe/autoware.universe/perception/autoware_tensorrt_common/src/tensorrt_common.cpp:536:28: error: operator '<' has no left operand
  536 | #if TENSORRT_VERSION_MAJOR < 8
      |                            ^
/home/dekor-x/autoware/src/universe/autoware.universe/perception/autoware_tensorrt_common/src/tensorrt_common.cpp: In member function ‘bool autoware::tensorrt_common::TrtCommon::buildEngineFromOnnx(const string&, const string&)’:
/home/dekor-x/autoware/src/universe/autoware.universe/perception/autoware_tensorrt_common/src/tensorrt_common.cpp:539:45: error: ‘plan’ was not declared in this scope
  539 |   file.write(reinterpret_cast<const char *>(plan->data()), plan->size());
      |                                             ^~~~
gmake[2]: *** [CMakeFiles/autoware_tensorrt_common.dir/build.make:76: CMakeFiles/autoware_tensorrt_common.dir/src/tensorrt_common.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/autoware_tensorrt_common.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< autoware_tensorrt_common [21.0s, exited with code 2]
```

### Steps to reproduce

1. Install [TensorRT 8.6 GA from Nvidia](https://developer.nvidia.com/nvidia-tensorrt-8x-download) (it shoudln't make lots of difference from the installation tutorial, if at all.)
2. Create symbolic links to the .so-files of TensorRT, so cmake can find the files and headers.
3. Try and build autoware according to the source installation tutorial

If i need to go more into detail, please let me know.

### Versions

- OS: Ubuntu 22.04.5 LTS
- ROS2: humble
- Autoware: main-branch (as of 2024.12.12)

### Possible causes

TensorRT defines the variables for the version numbers with a different name than Autoware/Cmake expects. Compare the NvInferVersion.h to the error output at **Actual Behavior**.

NvInferVersion.h:
```
/*
 * SPDX-FileCopyrightText: Copyright (c) 1993-2023 NVIDIA CORPORATION & AFFILIATES. All rights reserved.
 * SPDX-License-Identifier: LicenseRef-NvidiaProprietary
 *
 * NVIDIA CORPORATION, its affiliates and licensors retain all intellectual
 * property and proprietary rights in and to this material, related
 * documentation and any modifications thereto. Any use, reproduction,
 * disclosure or distribution of this material and related documentation
 * without an express license agreement from NVIDIA CORPORATION or
 * its affiliates is strictly prohibited.
 */

//!
//! \file NvInferVersion.h
//!
//! Defines the TensorRT version
//!
#ifndef NV_INFER_VERSION_H
#define NV_INFER_VERSION_H

#define NV_TENSORRT_MAJOR 8 //!< TensorRT major version.
#define NV_TENSORRT_MINOR 6 //!< TensorRT minor version.
#define NV_TENSORRT_PATCH 1 //!< TensorRT patch version.
#define NV_TENSORRT_BUILD 6 //!< TensorRT build number.

#define NV_TENSORRT_LWS_MAJOR 0 //!< TensorRT LWS major version.
#define NV_TENSORRT_LWS_MINOR 0 //!< TensorRT LWS minor version.
#define NV_TENSORRT_LWS_PATCH 0 //!< TensorRT LWS patch version.

// This #define is deprecated in TensorRT 8.6 and will be removed in 10.0. Use NV_TENSORRT_MAJOR.
#define NV_TENSORRT_SONAME_MAJOR 8 //!< Shared object library major version number.
// This #define is deprecated in TensorRT 8.6 and will be removed in 10.0. Use NV_TENSORRT_MINOR.
#define NV_TENSORRT_SONAME_MINOR 6 //!< Shared object library minor version number.
// This #define is deprecated in TensorRT 8.6 and will be removed in 10.0. Use NV_TENSORRT_PATCH.
#define NV_TENSORRT_SONAME_PATCH 1 //!< Shared object library patch version number.

#define NV_TENSORRT_RELEASE_TYPE_EARLY_ACCESS 0         //!< An early access release
#define NV_TENSORRT_RELEASE_TYPE_RELEASE_CANDIDATE 1    //!< A release candidate
#define NV_TENSORRT_RELEASE_TYPE_GENERAL_AVAILABILITY 2 //!< A final release

#define NV_TENSORRT_RELEASE_TYPE NV_TENSORRT_RELEASE_TYPE_GENERAL_AVAILABILITY //!< TensorRT release type

#endif // NV_INFER_VERSION_H
```

amadeuszsz, 2024-12-12:
Macros are placed [somewhere else](https://github.com/tier4/tensorrt_cmake_module/blob/main/cmake/Modules/FindTENSORRT.cmake). Please, use rosdep and build package once again.

vollkornholztuer, 2024-12-13:
Am i supposed to change something in my .bashrc or anywhere else?

vollkornholztuer, 2024-12-13:
Am i supposed to change something in my .bashrc or anywhere else? From reading the linked CMakeLists.txt, I think I have to define `TENSORRT_ROOT=/usr` in my .bashrc, is that correct? - I'm sorry for not knowing better.

As of now, i managed to build it correctly by changing the names of the macros inside the condition-checks, even if it's not the intended way. - I'd rather have it built from the original code though.

amadeuszsz, 2024-12-15:
This is default path where TRT is located. If it's same for you, you don't need to define this environment variable. It seems `TENSORRT_INCLUDE_DIR` is not set in makefile and I don't know why. Possibly because of your custom environment. You can:
1. Install TensorRT via apt CLI `sudo apt install libnvinfer8`. It is recommended way and it is automatically triggered if you follow our installation instruction (`setup-dev-env.sh` script).  
2. Use our docker images.
3. Troubleshoot the source of issue:
* Create CMakeLists.txt with content:
```cmake
cmake_minimum_required(VERSION 3.14)
project(trt_test)

find_package(tensorrt_cmake_module REQUIRED)
find_package(TENSORRT)
```
* Build it with `cmake .. --trace-expand &> log.txt`
* Follow the logs and understand the problem. In my case, I can see in line 970 of log
```
-- Found TensorRT headers at /usr/include/x86_64-linux-gnu
```

shuofeng1, 2024-12-16:
> Am i supposed to change something in my .bashrc or anywhere else? From reading the linked CMakeLists.txt, I think I have to define `TENSORRT_ROOT=/usr` in my .bashrc, is that correct? - I'm sorry for not knowing better.我应该更改我的 .bashrc 或其他任何地方的内容吗？通过阅读链接CMakeLists.txt，我认为我必须在我的 .bashrc 中定义 `TENSORRT_ROOT=/usr` ，对吗？- 很抱歉没有更好地了解。
> 
> As of now, i managed to build it correctly by changing the names of the macros inside the condition-checks, even if it's not the intended way. - I can revert these changes and try to rebuild it, because I'd also rather have the package build from the source code.截至目前，我设法通过在条件检查中更改宏的名称来正确构建它，即使这不是预期的方式。- 我可以撤销这些更改并尝试重新构建它，因为我也宁愿从源代码构建包。

Set TENSORRT_ROOT to your installation path in .bashrc 
Please refer to: https://github.com/autowarefoundation/autoware/issues/5523#issuecomment-2515996565

closed 2025-02-14 by amadeuszsz