autowarefoundation/autoware #5568: Autoware planning simulator not working

siddharth-w, 2024-12-16:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

[component_container_mt-27] [INFO] [1734273416.566069339] [route_handler]: getMainLanelets: lanelet_sequence = [ids: 7 ]
[component_container_mt-27] [INFO] [1734273416.585345760] [planning.mission_planning.mission_planner]: initial pose - x: 18.770046, y: 2.774801, z: -2.949700
[component_container_mt-27] [INFO] [1734273416.585415237] [planning.mission_planning.mission_planner]: initial orientation - qx: -0.000688, qy: 0.011501, qz: 0.059675, qw: 0.998151
[component_container_mt-27] [INFO] [1734273416.585438313] [planning.mission_planning.mission_planner]: goal pose - x: 60.170044, y: 5.674801, z: 0.000000
[component_container_mt-27] [INFO] [1734273416.585455251] [planning.mission_planning.mission_planner]: goal orientation - qx: 0.000000, qy: 0.000000, qz: 0.053342, qw: 0.998576
[component_container_mt-38] [INFO] [1734273416.669294439] [control.trajectory_follower.lane_departure_checker_node]: waiting for reference_trajectory msg...
[topic_state_monitor_node-11] [INFO] [1734273417.093917596] [system.topic_state_monitor_scenario_planning_trajectory]: /planning/scenario_planning/trajectory has not received. Set ERROR in diagnostics.
[topic_state_monitor_node-12] [INFO] [1734273417.156822224] [system.topic_state_monitor_trajectory_follower_control_cmd]: /control/trajectory_follower/control_cmd has not received. Set ERROR in diagnostics.
[component_container_mt-38] [INFO] [1734273417.767180644] [control.trajectory_follower.lane_departure_checker_node]: waiting for predicted_trajectory msg...
**[component_container_mt-38] component_container_mt: /usr/include/eigen3/Eigen/src/Core/CwiseBinaryOp.h:116: Eigen::CwiseBinaryOp<BinaryOp, Lhs, Rhs>::CwiseBinaryOp(const Lhs&, const Rhs&, const BinaryOp&) [with BinaryOp = Eigen::internal::scalar_sum_op<double, double>; LhsType = const Eigen::Matrix<double, -1, 1>; RhsType = const Eigen::CwiseBinaryOp<Eigen::internal::scalar_product_op<double, double>, const Eigen::Matrix<double, -1, 1>, const Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, const Eigen::Matrix<double, -1, 1> > >; Eigen::CwiseBinaryOp<BinaryOp, Lhs, Rhs>::Lhs = Eigen::Matrix<double, -1, 1>; Eigen::CwiseBinaryOp<BinaryOp, Lhs, Rhs>::Rhs = Eigen::CwiseBinaryOp<Eigen::internal::scalar_product_op<double, double>, const Eigen::Matrix<double, -1, 1>, const Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, const Eigen::Matrix<double, -1, 1> > >]: Assertion `aLhs.rows() == aRhs.rows() && aLhs.cols() == aRhs.cols()' failed.**
[component_container_mt-38] *** Aborted at 1734273417 (unix time) try "date -d @1734273417" if you are using GNU date ***
[component_container_mt-38] PC: @                0x0 (unknown)
[component_container_mt-38] *** SIGABRT (@0x3e8000262eb) received by PID 156395 (TID 0x7a7059400640) from PID 156395; stack trace: ***
[component_container_mt-38]     @     0x7a706048c520 (unknown)
[component_container_mt-38]     @     0x7a70604e09fc pthread_kill
[component_container_mt-38]     @     0x7a706048c476 raise
[component_container_mt-38]     @     0x7a70604727f3 abort
[topic_state_monitor_node-11] [WARN] [1734273418.026934594] [system.topic_state_monitor_scenario_planning_trajectory]: /planning/scenario_planning/trajectory topic rate has dropped to the warning level. Set WARN in diagnostics.
[component_container_mt-38]     @     0x7a706047271b (unknown)
[component_container_mt-38]     @     0x7a7060483e96 __assert_fail
[component_container_mt-38]     @     0x7a705306a974 Eigen::CwiseBinaryOp<>::CwiseBinaryOp()
[component_container_mt-38]     @     0x7a705306a40c Eigen::MatrixBase<>::operator+<>()
[component_container_mt-38]     @     0x7a705306ddde _ZZNK8autoware6motion7control22mpc_lateral_controller22KinematicsBicycleModel45calculatePredictedTrajectoryInWorldCoordinateERKN5Eigen6MatrixIdLin1ELin1ELi0ELin1ELin1EEES8_S8_S8_S8_S8_RKNS2_13MPCTrajectoryEdENKUlRKNS5_IdLin1ELi1ELi0ELin1ELi1EEERKdddE0_clESE_SG_dd
[component_container_mt-38]     @     0x7a705306dfa3 autoware::motion::control::mpc_lateral_controller::KinematicsBicycleModel::calculatePredictedTrajectoryInWorldCoordinate()
[component_container_mt-38]     @     0x7a7052fd7c29 autoware::motion::control::mpc_lateral_controller::MPC::calculatePredictedTrajectory()
[component_container_mt-38]     @     0x7a7052fd16d0 autoware::motion::control::mpc_lateral_controller::MPC::calculateMPC()
[component_container_mt-38]     @     0x7a7052f70c7d autoware::motion::control::mpc_lateral_controller::MpcLateralController::run()
[component_container_mt-38]     @     0x7a7053b795d1 autoware::motion::control::trajectory_follower_node::Controller::callbackTimerControl()
[component_container_mt-38]     @     0x7a7053cf5a72 std::__invoke_impl<>()
[component_container_mt-38]     @     0x7a7053cec4ff std::__invoke<>()
[component_container_mt-38]     @     0x7a7053ce1d8f _ZNSt5_BindIFMN8autoware6motion7control24trajectory_follower_node10ControllerEFvvEPS4_EE6__callIvJEJLm0EEEET_OSt5tupleIJDpT0_EESt12_Index_tupleIJXspT1_EEE
[component_container_mt-38]     @     0x7a7053cc57c7 std::_Bind<>::operator()<>()
[component_container_mt-38]     @     0x7a7053cbf150 rclcpp::GenericTimer<>::execute_callback_delegate<>()
[component_container_mt-38]     @     0x7a7053cb4857 rclcpp::GenericTimer<>::execute_callback()
[component_container_mt-38]     @     0x7a70609dafd1 rclcpp::Executor::execute_any_executable()
[component_container_mt-38]     @     0x7a70609e23ba rclcpp::executors::MultiThreadedExecutor::run()
[component_container_mt-38]     @     0x7a7060771253 (unknown)
[component_container_mt-38]     @     0x7a70604deac3 (unknown)
[component_container_mt-38]     @     0x7a706056fa04 clone
[topic_state_monitor_node-13] [INFO] [1734273419.360233946] [system.topic_state_monitor_control_command_control_cmd]: /control/command/control_cmd has not received. Set ERROR in diagnostics.
[ERROR] [component_container_mt-38]: process has died [pid 156395, exit code -6, cmd '/opt/ros/humble/lib/rclcpp_components/component_container_mt --ros-args -r __node:=control_container -r __ns:=/control -p use_sim_time:=False -p wheel_radius:=0.383 -p wheel_width:=0.235 -p wheel_base:=2.79 -p wheel_tread:=1.64 -p front_overhang:=1.0 -p rear_overhang:=1.1 -p left_overhang:=0.128 -p right_overhang:=0.128 -p vehicle_height:=2.5 -p max_steer_angle:=0.7'].

### Expected behavior

After setting up goal position, the vehicle should move towards the goal.

### Actual behavior

After setting up 2D Goal Position, the Auto button is greyed out and I get this error for dimension mismatch in mpc controller

### Steps to reproduce

1. Launch Planning Simulator
2. Setup pose estimate and 2D goal pose through Rviz GUI

### Versions

-OS: Ubuntu 22.04
-ROS2: Humble
-Autoware: Main repo/0.39.0

maxime-clem, 2024-12-16:
Did this issue appear on a fresh install, or after an update of your workspace ?

siddharth-w, 2024-12-16:
I encountered this on a fresh install, and tried with different releases/tags but no luck.

maxime-clem, 2024-12-16:
Did you setup your environment using the `setup-dev-env.sh` script ? The issue may be caused by incorrect library versions.

Omar-Mabrouk-9, 2025-01-19:
@siddharth-w can you tell me how did you fix the issue?

siddharth-w, 2025-01-19:
@Omar-Mabrouk-9  I couldn't solve it, I think I closed the issue by mistake. I did shift to working with the docker container but its working fine with any other system. So, its most probably due to some library version mismatch in the planning module(which I wa too busy to sit and pin-point) and a clean reinstall of everything should fix it. 
Or but its some hardware limitation issue as that particular system I was using was an Industrial PC with 4 cores :p (Idk, I am tech illiterate in hardware)

Imesh-VEGA, 2025-02-06:
Hi

I'm experiencing the same issue on a fresh installation of Autoware. Initially, I was able to run the simulator with both the Autoware runtime container and my local development workspace, but a few days ago, I started encountering this error. Now, even after setting up a completely new Ubuntu environment and creating a fresh Autoware development workspace, the issue persists.
Development Setup:

    Hardware: Jetson AGX Orin
    OS: Ubuntu 22.04.4 LTS
    ROS Version: ROS 2 Humble
    Installation Method: ADE container with local development workspace

I Would appreciate any insights from the community or maintainers regarding this issue. Have you found a workaround or identified any potential causes?

Thanks!

maxime-clem, 2025-02-13:
Since the issue cannot be reproduced it is difficult to investigate and understand the cause.
If you are able to reproduce the crash after building with debug flags and either generating a core dump or analyzing the issue yourself with `gdb`, that would be really helpful.

evshary, 2025-03-11:
We faced the same issue, but it can be solved by applying [the PR](https://github.com/autowarefoundation/autoware.universe/pull/10235).
Or you can just build in release mode.