autowarefoundation/autoware #5607: CUDA, TensorRT, CUDNN upgrade

amadeuszsz, 2024-12-24:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

See related [discussion](https://github.com/orgs/autowarefoundation/discussions/5300) 

### Purpose

Incoming new packages requires TensorRT upgrade.

### Possible approaches

1. Refactor packages which uses TensorRT in autoware.universe and keep compatibility in current environment.
2. Wait for merge [rosdistro](https://github.com/ros/rosdistro/pull/44031) PR. If fast integration needed, we can add [tensorrt_cmake_module](https://github.com/tier4/tensorrt_cmake_module) to `autoware.repos` temporary.
3. Merge PR for autoware.universe.
4. Merge PR for CUDA, CUDNN, TensorRT upgrade in Autoware.

### Definition of done

- [ ] Update `tensorrt_cmake_module`.
- [ ] Refactor `autoware_tensorrt_common`.
- [ ] Refactor related packages.
  - [ ] `autoware_image_projection_based_fusion`.
  - [ ] `autoware_lidar_apollo_instance_segmentation`.
  - [ ] `autoware_lidar_centerpoint`.
  - [ ] `autoware_lidar_transfusion`.
  - [ ] `autoware_shape_estimation`.
  - [ ] `autoware_tensorrt_classifier`.
  - [ ] `autoware_tensorrt_yolox`.
  - [ ] `autoware_traffic_light_classifier`.
  - [ ] `autoware_traffic_light_fine_detector`.
  - [ ] `autoware_tensorrt_rtmdet`.
- [ ] Upgrade CUDA, CUDNN, TensorRT for Autoware.
- [ ] Wait for rosdistro merge.
- [ ] Merge autoware.universe PR.
- [ ] Merge autoware PR.

razr, 2024-12-28:
Hi @amadeuszsz, I think I have a related question.

I have the following error, when I try to run a Carla Autoware example inside the `ghcr.io/autowarefoundation/autoware:universe-cuda` docker container

```bash
[component_container_mt-61] [ERROR] [1735397635.819246109] [perception.traffic_light_recognition.traffic_light.classification.car_traffic_light_classifier]: please install CUDA, CUDNN and TensorRT to use cnn classifier
```

I run this example

```bash
ros2 launch autoware_launch e2e_simulator.launch.xml map_path:=/tmp/autoware_map/Town01 \
  vehicle_model:=sample_vehicle sensor_model:=awsim_sensor_kit simulator_type:=carla carla_map:=Town01
```

and my docker is

````bash
```

amadeuszsz, 2025-01-06:
@razr your image contains TensorRT, you can check it by `dpkg -l | grep nvinfer`.
You can also check if libraries are linked properly:
```bash
ldd /opt/autoware/lib/libtraffic_light_classifier_nodelet.so | grep libnv
        libnvinfer.so.8 => /usr/lib/x86_64-linux-gnu/libnvinfer.so.8 (0x00007808ea97c000)
        libnvinfer_plugin.so.8 => /usr/lib/x86_64-linux-gnu/libnvinfer_plugin.so.8 (0x00007808e8500000)
        libnvonnxparser.so.8 => /usr/lib/x86_64-linux-gnu/libnvonnxparser.so.8 (0x00007808e8000000)
```

or if you use locally built workspace:
```bash
ldd /workspace/install/autoware_traffic_light_classifier/lib/libtraffic_light_classifier_nodelet.so | grep libnv                                                                                                                          
        libnvinfer.so.8 => /usr/lib/x86_64-linux-gnu/libnvinfer.so.8 (0x000074245a897000)                                                                                                                                                                                          
        libnvinfer_plugin.so.8 => /usr/lib/x86_64-linux-gnu/libnvinfer_plugin.so.8 (0x000074245841b000)                                                                                                                                                                            
        libnvonnxparser.so.8 => /usr/lib/x86_64-linux-gnu/libnvonnxparser.so.8 (0x0000742457e00000) 
```

If you use locally build workspace, I suggest to make a clean build and make sure you use CUDA-supported image.

Please create a separate thread if you issue still occurs.

alperenyksl, 2025-01-29:
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 570.86.10              Driver Version: 570.86.10      CUDA Version: 12.8     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 3060 ...    Off |   00000000:01:00.0 Off |                  N/A |
| N/A   41C    P0             24W /   80W |      15MiB /   6144MiB |      8%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI              PID   Type   Process name                        GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A            1192      G   /usr/lib/xorg/Xorg                        4MiB |
+-----------------------------------------------------------------------------------------+
I had a problem similar to yours, after that ı reset my ubuntu and download 22.04, downloaded nvidia driver 570 and install CUDA 12.8 in https://developer.nvidia.com/cuda-downloads?target_os=Linux&target_arch=x86_64&Distribution=Ubuntu&target_version=24.04&target_type=deb_network.
I hope you will solve this with settings.

amadeuszsz, 2025-02-21:
Confirmed recently deployed Docker images consist of upgraded dependencies.

closed 2025-02-21 by amadeuszsz