autowarefoundation/autoware #5659: Loacalization failure - ndt_scan_matcher

siddharth-w, 2025-01-15:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Error while doing 2D pose estimate with:
```[autoware_pose_initializer_node-46] [INFO] [1736947007.374818653] [localization.util.pose_initializer]: EKF triggering service is available!
[autoware_pose_initializer_node-46] [INFO] [1736947007.375356377] [localization.util.pose_initializer]: EKF Deactivation succeeded
[autoware_pose_initializer_node-46] [INFO] [1736947007.375514676] [localization.util.pose_initializer]: NDT triggering service is available!
[autoware_pose_initializer_node-46] [INFO] [1736947007.375849799] [localization.util.pose_initializer]: NDT Deactivation succeeded
[autoware_pose_initializer_node-46] [INFO] [1736947007.375942822] [localization.util.pose_initializer]: Call align server.
[topic_state_monitor_node-13] [INFO] [1736947007.396806086] [system.topic_state_monitor_initialpose3d]: /initialpose3d has not received. Set ERROR in diagnostics.
[topic_state_monitor_node-25] [INFO] [1736947007.412860535] [system.topic_state_monitor_transform_map_to_base_link]: /tf has not received. Set ERROR in diagnostics.
[gnss_poser-42] [ERROR] [1736947007.521712299] [sensing.gnss.gnss_poser]: map_projector_info is local projector type. Unable to convert GNSS pose.
[topic_state_monitor_node-14] [INFO] [1736947007.615482878] [system.topic_state_monitor_pose_twist_fusion_filter_pose]: /localization/pose_twist_fusion_filter/pose has not received. Set ERROR in diagnostics.
[topic_state_monitor_node-21] [INFO] [1736947007.653593006] [system.topic_state_monitor_control_command_control_cmd]: /control/command/control_cmd has not received. Set ERROR in diagnostics.
[topic_state_monitor_node-23] [INFO] [1736947007.679504890] [system.topic_state_monitor_vehicle_status_steering_status]: /vehicle/status/steering_status has not received. Set ERROR in diagnostics.
[topic_state_monitor_node-17] [INFO] [1736947007.693482355] [system.topic_state_monitor_traffic_light_recognition_traffic_signals]: /perception/traffic_light_recognition/traffic_signals has not received. Set ERROR in diagnostics.
[topic_state_monitor_node-19] [INFO] [1736947007.751435257] [system.topic_state_monitor_scenario_planning_trajectory]: /planning/scenario_planning/trajectory has not received. Set ERROR in diagnostics.
[autoware_ndt_scan_matcher_node-44] [WARN] [1736947007.889229266] [localization.pose_estimator.ndt_scan_matcher]: No InputSource. Please check the input lidar topic
[service_log_checker_node-9] [ERROR] [1736947007.889894503] [system.service_log_checker]: /localization/initialize: status code 4 'align server failed.' (/localization/util/pose_initializer)
[service_log_checker_node-9] [ERROR] [1736947007.890569301] [system.service_log_checker]: /localization/initialize: status code 4 'align server failed.' (/adapi/node/localization)
[service_log_checker_node-9] [ERROR] [1736947007.890644477] [system.service_log_checker]: /api/localization/initialize: status code 4 'align server failed.' (/adapi/node/localization)
[service_log_checker_node-9] [ERROR] [1736947007.890693512] [system.service_log_checker]: /api/localization/initialize: status code 4 'align server failed.' (/default_ad_api/helpers/initial_pose_adaptor)
[autoware_ekf_localizer_node-47] [WARN] [1736947008.102694164] [localization.pose_twist_fusion_filter.ekf_localizer]: The node is not activated. Provide initial pose to pose_initializer

```

### Expected behavior

Localizes and vehicle appears in Rviz

### Actual behavior

The NDT input source:/localization/util/downsample/pointcloud **is getting published but with huge latency**
```
~/autoware_src$ ros2 topic  info -v /localization/util/downsample/pointcloud
Type: sensor_msgs/msg/PointCloud2

Publisher count: 1

Node name: random_downsample_filter
Node namespace: /localization/util
Topic type: sensor_msgs/msg/PointCloud2
Endpoint type: PUBLISHER
GID: 01.0f.4e.9a.b3.60.18.01.00.00.00.00.00.00.5b.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: BEST_EFFORT
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 1

Node name: ndt_scan_matcher
Node namespace: /localization/pose_estimator
Topic type: sensor_msgs/msg/PointCloud2
Endpoint type: SUBSCRIPTION
GID: 01.0f.4e.9a.cf.60.c7.33.00.00.00.00.00.00.18.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: BEST_EFFORT
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

```
and _/debug/published_time_ **shows huge latency**
```
~/autoware_src$ ros2 topic echo /localization/util/downsample/pointcloud/debug/published_time
header:
  stamp:
    sec: 1736946382
    nanosec: 970611968
  frame_id: ''
published_stamp:
  sec: 1736946395
  nanosec: 255266780
---
header:
  stamp:
    sec: 1736946386
    nanosec: 771005696
  frame_id: ''
published_stamp:
  sec: 1736946398
  nanosec: 444278824
---
header:
  stamp:
    sec: 1736946388
    nanosec: 771509248
  frame_id: ''
published_stamp:
  sec: 1736946404
  nanosec: 596483621

```

The topic /initialpose3d does not echo anything when we do 2D pose estimate in rviz:
```~/autoware_src$ ros2 topic info -v /initialpose3d 
Type: geometry_msgs/msg/PoseWithCovarianceStamped

Publisher count: 1

Node name: pose_initializer
Node namespace: /localization/util
Topic type: geometry_msgs/msg/PoseWithCovarianceStamped
Endpoint type: PUBLISHER
GID: 01.0f.4e.9a.d3.60.8f.96.00.00.00.00.00.00.16.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 1

Node name: ekf_localizer
Node namespace: /localization/pose_twist_fusion_filter
Topic type: geometry_msgs/msg/PoseWithCovarianceStamped
Endpoint type: SUBSCRIPTION
GID: 01.0f.4e.9a.d7.60.7f.19.00.00.00.00.00.00.20.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

```

All the crop_box_filter nodes have huge publishing latency even though processing time is less:
```
~/autoware_src$ ros2 topic echo /sensing/lidar/top/crop_box_filter_self/debug/processing_time_ms 
stamp:
  sec: 1736945884
  nanosec: 200993696
data: 0.081
---
stamp:
  sec: 1736945887
  nanosec: 261394376
data: 0.071
---
stamp:
  sec: 1736945892
  nanosec: 237265857
data: 0.061
---
stamp:
  sec: 1736945908
  nanosec: 937567316
data: 0.062
---
```
```
~/autoware_src$ ros2 topic echo /sensing/lidar/top/crop_box_filter_self/debug/pipeline_latency_ms
stamp:
  sec: 1736945918
  nanosec: 613404701
data: 542.600434
---
stamp:
  sec: 1736945920
  nanosec: 590051031
data: 518.517973
---
stamp:
  sec: 1736945924
  nanosec: 439817692
data: 569.361814
---

```

### Steps to reproduce

1. Launch autoware.launch.xml
2. In Rviz do 2D pose estimate

### Versions

**OS: Ubuntu 22.04.5 LTS x86_64 
CPU: Intel Xeon W-1390P (16) @ 5.100GHz 
GPU: NVIDIA RTX A5000 
GPU: Intel RocketLake-S GT1 [UHD Graphics P750] 
Memory: 7548MiB / 64026MiB** 

### Additional context

VLP16 Lidar
ublox gps
zed2 camera and IMU

YamatoAndo, 2025-01-16:
@siddharth-w Hi.
Could you run the following command to launch `rqt_runtime_monitor` and check [the diagnostic information for ndt_scan_matcher](https://github.com/autowarefoundation/autoware.universe/tree/main/localization/autoware_ndt_scan_matcher#diagnostics)?
```
ros2 run rqt_runtime_monitor rqt_runtime_monitor
```
If what you’re saying is correct, there might be an error at `sensor_points_delay_time_sec`.
If so, please adjust [sensor_points.timeout_sec](https://github.com/autowarefoundation/autoware_launch/blob/main/autoware_launch/config/localization/ndt_scan_matcher/ndt_scan_matcher.param.yaml#L15-L16).

siddharth-w, 2025-01-16:
There was an error indeed 
[image]

But even after setting the timeout_sec to a very high value (20, 50, 300), I'd still get the same error of No Input, and I am sure the input msg is getting published:
```
ros2 topic echo /localization/util/downsample/pointcloud/debug/published_time 
header:
  stamp:
    sec: 1737020492
    nanosec: 64168192
  frame_id: ''
published_stamp:
  sec: 1737020503
  nanosec: 59771265
---
header:
  stamp:
    sec: 1737020494
    nanosec: 64718592
  frame_id: ''
published_stamp:
  sec: 1737020525
  nanosec: 609161645
---
```

One more thing, the "input_regularization_pose_topic" value="/sensing/gnss/pose_with_covariance" is not getting published as I have no clue what to do for the **/autoware_orientation** topic in **gnss_poser** node. Can that be an issue too other than the pointcloud input timeout ?

Whats causing the latency in pointcloud pipeline ? It shouldn't be the system capabilities in my case, the processing time is very less but pipeline latency is extremely high.
Also, what might be the issue for initialpoint3d not getting published ? 

Can I get some elaborated help please ? :(

YamatoAndo, 2025-01-17:
It seems that there is also an error `sensor_points_max_distance`. Please adjust [sensor_points.required_distance](https://github.com/autowarefoundation/autoware_launch/blob/main/autoware_launch/config/localization/ndt_scan_matcher/ndt_scan_matcher.param.yaml#L20).

> One more thing, the "input_regularization_pose_topic" value="/sensing/gnss/pose_with_covariance" is not getting published as I have no clue what to do for the /autoware_orientation topic in gnss_poser node. Can that be an issue too other than the pointcloud input timeout ?

This issue is unrelated to the current problem, so you don't need to worry about it.
The "input_regularization_pose_topic" is optional and will function properly even if it's not correctly configured.

> Whats causing the latency in pointcloud pipeline ? It shouldn't be the system capabilities in my case, the processing time is very less but pipeline latency is extremely high.

There isn't enough information to determine the cause.
Discussing this topic here might make the thread too long, so please create a new issue and ask someone else for assistance.

> Also, what might be the issue for initialpoint3d not getting published ?

If this issue is not resolved, initialpoint3d will not be published.

siddharth-w, 2025-01-17:
Changing [sensor_points.required_distance](https://github.com/autowarefoundation/autoware_launch/blob/main/autoware_launch/config/localization/ndt_scan_matcher/ndt_scan_matcher.param.yaml#L20) solved the issue, Thanks!

I'll close the issue and start a different one for the pointcloud pipeline issue.
Thanks a lot! :)

closed 2025-01-17 by siddharth-w