autowarefoundation/autoware #5759: I can't run awsim in docker

kimsos0723, 2025-02-11:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I'm convinced that this is not my fault but a bug.

### Description

[image]

### Expected behavior

It should run successfully

### Actual behavior

[image]

### Steps to reproduce

1. git clone autoware
2. ./setup-dev-en.sh docker
3. ./docker/run.sh --devel
4. mkdir src
5. vcs import src < autoware.repos
6. sudo apt update && upgrade
7. source /opt/ros/humble/setup.bash
rosdep update
rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO
8. colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release
9. ./docker/run.sh --map-path /home/swkim/Downloads/nishishinjuku_autoware_map ros2 launch autoware_launch e2e_simulator.launch.xml vehicle_model:=awsim_labs_vehicle sensor_model:=awsim_labs_sensor_kit map_path:=/home/swkim/Downloads/nishishinjuku_autoware_map launch_vehicle_interface:=true

kimsos0723, 2025-02-12:
This issue seems to be related to https://github.com/autowarefoundation/AWSIM-Labs/issues/188 
so i'll try it fist.

xmfcx, 2025-02-17:
Yes, Autoware has been failing with AWSIM Labs for a while due to some launch misconfiguration.

- https://github.com/autowarefoundation/awsim_labs_sensor_kit_launch/pull/9
- https://github.com/autowarefoundation/awsim_labs_sensor_kit_launch/pull/22

are merged, and I've tested AWSIM Labs with these changes merged. Could you try again with the latest main branch?

kimsos0723, 2025-02-19:
[image]
still same error occur ....

xmfcx, 2025-02-19:
New changes are probably not reflected to the docker image yet.

At your 5th stage, `vcs import src < autoware.repos`, after doing this, can you also run: `vcs import src < autoware-nightly.repos` and `vcs pull src`?

Also `rm -rf build install log` then compile again.

kimsos0723, 2025-02-19:
[image]
oh, another errors occured