autowarefoundation/autoware #5783: run simulation error

moonseoppkim, 2025-02-14:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I'm convinced that this is not my fault but a bug.

### Description

Hi, I want to run some simulations with the latest docker image(autoware.universe) without cuda.
But, it doesn't work as "package 'autoware_probabilistic_occupancy_grid_map' not found" this error msg.

I refered "https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/docker-installation/" this document.

1. Docker image download
seop@seop-virtual-machine:~/autoware$ docker images
REPOSITORY                            TAG              IMAGE ID       CREATED        SIZE
ghcr.io/autowarefoundation/autoware   universe         439f6e0706d6   18 hours ago   5.01GB
(no nvidia)

2. ./setup-dev-env.sh -y --no-nvidia docker
success, without any issues

3. ./docker/run.sh --no-nvidia --map-path /home/seop/autoware_map/sample-map-planning

seop@seop-virtual-machine:~/autoware_map/sample-map-planning$ ls
lanelet2_map.osm  map_config.yaml  map_projector_info.yaml  pointcloud_map.pcd
seop@seop-virtual-machine:~/autoware_map/sample-map-planning$ ls -al
total 28532
drwxrwxr-x 2 seop seop     4096 Nov  8  2023 .
drwxrwxr-x 4 seop seop     4096 Feb 13 10:02 ..
-rw-rw-r-- 1 seop seop  1071550 Nov  8  2023 lanelet2_map.osm
-rw-rw-r-- 1 seop seop      178 Nov  8  2023 map_config.yaml
-rw-rw-r-- 1 seop seop       60 Nov  8  2023 map_projector_info.yaml
-rw-rw-r-- 1 seop seop 28125828 Nov  8  2023 pointcloud_map.pcd

4. Error msg
seop@seop-virtual-machine:~/autoware$ ./docker/run.sh --no-nvidia --map-path /home/seop/autoware_map/sample-map-planning ros2 launch autoware_launch planning_simulator.launch.xml map_path:=/autoware_map vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit
-----------------------------------------------------------------
Launching Autoware
IMAGE: ghcr.io/autowarefoundation/autoware:universe
MAP PATH(mounted): /home/seop/autoware_map/sample-map-planning:/autoware_map
LAUNCH CMD: ros2 launch autoware_launch planning_simulator.launch.xml map_path:=/autoware_map vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit
-----------------------------------------------------------------
+ docker run -it --rm --net=host -e DISPLAY=:0 -v /tmp/.X11-unix/:/tmp/.X11-unix -e XAUTHORITY=/run/user/1000/.mutter-Xwaylandauth.UPET12 -e XDG_RUNTIME_DIR=/run/user/1000 -e NVIDIA_DRIVER_CAPABILITIES=all -v /etc/localtime:/etc/localtime:ro -v /home/seop/autoware_map/sample-map-planning:/autoware_map:ro ghcr.io/autowarefoundation/autoware:universe ros2 launch autoware_launch planning_simulator.launch.xml map_path:=/autoware_map vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit
[INFO] [launch]: All log files can be found below /root/.ros/log/2025-02-13-20-54-15-892881-seop-virtual-machine-1
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/pointcloud_container/glog_component' in container '/pointcloud_container'

****************************ERROR*****************************
****************************HERE*****************************
[ERROR] [launch]: Caught exception in launch (see debug for traceback): "package 'autoware_probabilistic_occupancy_grid_map' not found, searching: ['/opt/autoware', '/opt/ros/humble']"
****************************ERROR*****************************
****************************HERE*****************************

[WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the '/adapi/container/_container/load_node' service, due to shutdown.
[WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the '/autoware_api/external/rtc_controller/container/_container/load_node' service, due to shutdown.
[WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the '/control/control_container/_container/load_node' service, due to shutdown.
[WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the '/control/control_check_container/_container/load_node' service response, due to shutdown.
[WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the '/adapi/container/_container/load_node' service, due to shutdown.
Future exception was never retrieved
future: <Future finished exception=InvalidHandle('cannot use Destroyable because destruction was requested')>
Traceback (most recent call last):
  File "/usr/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch_ros/actions/load_composable_nodes.py", line 197, in _load_in_sequence
    self._load_node(next_load_node_request, context)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch_ros/actions/load_composable_nodes.py", line 119, in _load_node
    while not self.__rclpy_load_node_client.wait_for_service(timeout_sec=1.0):
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/client.py", line 176, in wait_for_service
    while self.context.ok() and not self.service_is_ready() and timeout_sec > 0.0:
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/client.py", line 159, in service_is_ready
    with self.handle:
rclpy._rclpy_pybind11.InvalidHandle: cannot use Destroyable because destruction was requested

I don't know why I can see this weird error msg.

Thanks.

### Expected behavior

./docker/run.sh --no-nvidia --map-path /home/seop/autoware_map/sample-map-planning

run this command perperly

### Actual behavior

[ERROR] [launch]: Caught exception in launch (see debug for traceback): "package 'autoware_probabilistic_occupancy_grid_map' not found, searching: ['/opt/autoware', '/opt/ros/humble']"

and 

Future exception was never retrieved
future: <Future finished exception=InvalidHandle('cannot use Destroyable because destruction was requested')>
Traceback (most recent call last):
  File "/usr/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch_ros/actions/load_composable_nodes.py", line 197, in _load_in_sequence
    self._load_node(next_load_node_request, context)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch_ros/actions/load_composable_nodes.py", line 119, in _load_node
    while not self.__rclpy_load_node_client.wait_for_service(timeout_sec=1.0):
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/client.py", line 176, in wait_for_service
    while self.context.ok() and not self.service_is_ready() and timeout_sec > 0.0:
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/client.py", line 159, in service_is_ready
    with self.handle:
rclpy._rclpy_pybind11.InvalidHandle: cannot use Destroyable because destruction was requested

### Steps to reproduce

1. Docker image download for non-cuda (autoware.universe)
2. ./setup-dev-env.sh -y --no-nvidia docker
3.  ./docker/run.sh --no-nvidia --map-path /home/seop/autoware_map/sample-map-planning

### Versions

- OS : ubuntu 22.04
- ROS : Ros2 humble
- Autoware : docker pull ghcr.io/autowarefoundation/autoware:universe-20250213

YuqiHuai, 2025-02-14:
+1 same issue on commit tag: 2025.02 with NVIDIA (`./setup-dev-env.sh -y docker`).

moonseoppkim, 2025-02-15:
> +1 same issue on commit tag: 2025.02 with NVIDIA (`./setup-dev-env.sh -y docker`).

Commits on Feb 12, 2025
[chore(autoware_probabilistic_occupancy_grid_map): fixed cuda use on non-cuda settings (](https://github.com/autowarefoundation/autoware.universe/commit/5d3b97a314c41c850700ac0b83c1c878598a116e)https://github.com/autowarefoundation/autoware.universe/pull/10099[)](https://github.com/autowarefoundation/autoware.universe/commit/5d3b97a314c41c850700ac0b83c1c878598a116e)
[knzo25](https://github.com/autowarefoundation/autoware.universe/commits?author=knzo25)
knzo25
authored
3 days ago
·
Verified
Commits on Feb 9, 2025
[fix(probabilistic_occupancy_grid_map): build and runtime on non-cuda environments (](https://github.com/autowarefoundation/autoware.universe/commit/4823597cd6c5f05725bb782dcadd8ce08e28df58)https://github.com/autowarefoundation/autoware.universe/pull/10061[)](https://github.com/autowarefoundation/autoware.universe/commit/4823597cd6c5f05725bb782dcadd8ce08e28df58)
[knzo25](https://github.com/autowarefoundation/autoware.universe/commits?author=knzo25)
knzo25
authored
5 days ago
·
Verified
[feat(autoware_probabilistic_occupancy_grid_map): improved data handling on the ogm (](https://github.com/autowarefoundation/autoware.universe/commit/c08e6b3bd19d9a8577a400b0930ad612090e581a)https://github.com/autowarefoundation/autoware.universe/pull/10060[)](https://github.com/autowarefoundation/autoware.universe/commit/c08e6b3bd19d9a8577a400b0930ad612090e581a)
[knzo25](https://github.com/autowarefoundation/autoware.universe/commits?author=knzo25)
knzo25
authored
5 days ago
·

I could pass this error with these commits.
(change to main branch not 0.41.1 branch)

ishit16, 2025-02-28:
> > +1 same issue on commit tag: 2025.02 with NVIDIA (`./setup-dev-env.sh -y docker`).
> 
> Commits on Feb 12, 2025 [chore(autoware_probabilistic_occupancy_grid_map): fixed cuda use on non-cuda settings (](https://github.com/autowarefoundation/autoware.universe/commit/5d3b97a314c41c850700ac0b83c1c878598a116e)[https://github.com/autowarefoundation/autoware.universe/pull/10099[)](https://github.com/autowarefoundation/autoware.universe/commit/5d3b97a314c41c850700ac0b83c1c878598a116e](https://github.com/autowarefoundation/autoware.universe/pull/10099%5B)%5D(https://github.com/autowarefoundation/autoware.universe/commit/5d3b97a314c41c850700ac0b83c1c878598a116e)) [knzo25](https://github.com/autowarefoundation/autoware.universe/commits?author=knzo25) knzo25 authored 3 days ago · Verified Commits on Feb 9, 2025 [fix(probabilistic_occupancy_grid_map): build and runtime on non-cuda environments (](https://github.com/autowarefoundation/autoware.universe/commit/4823597cd6c5f05725bb782dcadd8ce08e28df58)[https://github.com/autowarefoundation/autoware.universe/pull/10061[)](https://github.com/autowarefoundation/autoware.universe/commit/4823597cd6c5f05725bb782dcadd8ce08e28df58](https://github.com/autowarefoundation/autoware.universe/pull/10061%5B)%5D(https://github.com/autowarefoundation/autoware.universe/commit/4823597cd6c5f05725bb782dcadd8ce08e28df58)) [knzo25](https://github.com/autowarefoundation/autoware.universe/commits?author=knzo25) knzo25 authored 5 days ago · Verified [feat(autoware_probabilistic_occupancy_grid_map): improved data handling on the ogm (](https://github.com/autowarefoundation/autoware.universe/commit/c08e6b3bd19d9a8577a400b0930ad612090e581a)[https://github.com/autowarefoundation/autoware.universe/pull/10060[)](https://github.com/autowarefoundation/autoware.universe/commit/c08e6b3bd19d9a8577a400b0930ad612090e581a](https://github.com/autowarefoundation/autoware.universe/pull/10060%5B)%5D(https://github.com/autowarefoundation/autoware.universe/commit/c08e6b3bd19d9a8577a400b0930ad612090e581a)) [knzo25](https://github.com/autowarefoundation/autoware.universe/commits?author=knzo25) knzo25 authored 5 days ago ·
> 
> I could pass this error with these commits. (change to main branch not 0.41.1 branch)

Hey! I was also getting the same error, and I am still not able to resolve the error. 

## Some Context
I am trying to run Open AD Kit Containers and I was following this documentation. 
https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/docker-installation/

## Everything runs fine till this command 
./setup-dev-env.sh -y --no-nvidia docker

## After this when I try to run this command with the path to a sample map
./docker/run.sh --map-path path_to_map_data

## I get this error
+ docker run -it --rm --net=host -e XAUTHORITY= -e XDG_RUNTIME_DIR= -e NVIDIA_DRIVER_CAPABILITIES=all -v /etc/localtime:/etc/localtime:ro -v /root/autoware_map/sample-map-planning/:/autoware_map:ro ghcr.io/autowarefoundation/autoware:universe ros2 launch autoware_launch autoware.launch.xml map_path:=/autoware_map vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit

[INFO] [launch]: All log files can be found below /root/.ros/log/2025-02-28-11-48-47-446974-docker-desktop-1
[INFO] [launch]: Default logging verbosity is set to INFO
[WARNING] [launch_ros.actions.node]: Parameter file path is not a file: /opt/ros/humble/share/ublox_gps/c94_f9p_rover.yaml
[ERROR] [launch]: Caught exception in launch (see debug for traceback): "package 'autoware_probabilistic_occupancy_grid_map' not found, searching: ['/opt/autoware', '/opt/ros/humble']"
[WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the '/sensing/lidar/right/pointcloud_preprocessor/pointcloud_container/_container/load_node' service, due to shutdown.
[WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the '/map/map_container/_container/load_node' service, due to shutdown.
[WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the '/system/system_monitor/system_monitor/system_monitor_container/_container/load_node' service response, due to shutdown.
[WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the 'pointcloud_container/_container/load_node' service response, due to shutdown.
[WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the '/pointcloud_container/_container/load_node' service response, due to shutdown.
[WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the '/system/mrm_comfortable_stop_operator/mrm_comfortable_stop_operator_container/_container/load_node' service response, due to shutdown.
[WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the 'pointcloud_container/_container/load_node' service response, due to shutdown.
[WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the '/sensing/lidar/top/pointcloud_preprocessor/pointcloud_container/_container/load_node' service response, due to shutdown.
[WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the '/pointcloud_container/_container/load_node' service response, due to shutdown.
[WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the '/system/component_state_monitor/container/_container/load_node' service response, due to shutdown.
[... cut ...]
[WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the '/sensing/lidar/rear/pointcloud_preprocessor/pointcloud_container/_container/load_node' service response, due to shutdown.
Future exception was never retrieved

Now, I saw your comment where you mentioned that after those commits, you could resolve that error. But those commits are inside the autoware.universe repository right? But the documentation asks us to clone this repository
https://github.com/autowarefoundation/autoware.git

So how were you able to resolve the errors? I am on the latest pull of the repository mentioned above and in the main branch. Can you please elaborate on steps to resolve the error?

moonseoppkim, 2025-02-28:
> > > +1 same issue on commit tag: 2025.02 with NVIDIA (`./setup-dev-env.sh -y docker`).
> > 
> > 
> > Commits on Feb 12, 2025 [chore(autoware_probabilistic_occupancy_grid_map): fixed cuda use on non-cuda settings (](https://github.com/autowarefoundation/autoware.universe/commit/5d3b97a314c41c850700ac0b83c1c878598a116e)[https://github.com/[autowarefoundation/autoware.universe/pull/10099](https://github.com/autowarefoundation/autoware.universe/pull/10099)[)](https://github.com/autowarefoundation/autoware.universe/commit/5d3b97a314c41c850700ac0b83c1c878598a116e%5D(https://github.com/autowarefoundation/autoware.universe/pull/10099%5B)%5D(https://github.com/autowarefoundation/autoware.universe/commit/5d3b97a314c41c850700ac0b83c1c878598a116e)) [knzo25](https://github.com/autowarefoundation/autoware.universe/commits?author=knzo25) knzo25 authored 3 days ago · Verified Commits on Feb 9, 2025 [fix(probabilistic_occupancy_grid_map): build and runtime on non-cuda environments (](https://github.com/autowarefoundation/autoware.universe/commit/4823597cd6c5f05725bb782dcadd8ce08e28df58)[https://github.com/[autowarefoundation/autoware.universe/pull/10061](https://github.com/autowarefoundation/autoware.universe/pull/10061)[)](https://github.com/autowarefoundation/autoware.universe/commit/4823597cd6c5f05725bb782dcadd8ce08e28df58%5D(https://github.com/autowarefoundation/autoware.universe/pull/10061%5B)%5D(https://github.com/autowarefoundation/autoware.universe/commit/4823597cd6c5f05725bb782dcadd8ce08e28df58)) [knzo25](https://github.com/autowarefoundation/autoware.universe/commits?author=knzo25) knzo25 authored 5 days ago · Verified [feat(autoware_probabilistic_occupancy_grid_map): improved data handling on the ogm (](https://github.com/autowarefoundation/autoware.universe/commit/c08e6b3bd19d9a8577a400b0930ad612090e581a)[https://github.com/[autowarefoundation/autoware.universe/pull/10060](https://github.com/autowarefoundation/autoware.universe/pull/10060)[)](https://github.com/autowarefoundation/autoware.universe/commit/c08e6b3bd19d9a8577a400b0930ad612090e581a%5D(https://github.com/autowarefoundation/autoware.universe/pull/10060%5B)%5D(https://github.com/autowarefoundation/autoware.universe/commit/c08e6b3bd19d9a8577a400b0930ad612090e581a)) [knzo25](https://github.com/autowarefoundation/autoware.universe/commits?author=knzo25) knzo25 authored 5 days ago ·
> > I could pass this error with these commits. (change to main branch not 0.41.1 branch)
> 
> Hey! I was also getting the same error, and I am still not able to resolve the error.
> 
> ## Some Context
> I am trying to run Open AD Kit Containers and I was following this documentation. https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/docker-installation/
> 
> ## Everything runs fine till this command
> ./setup-dev-env.sh -y --no-nvidia docker
> 
> ## After this when I try to run this command with the path to a sample map
> ./docker/run.sh --map-path path_to_map_data
> 
> ## I get this error
> * docker run -it --rm --net=host -e XAUTHORITY= -e XDG_RUNTIME_DIR= -e NVIDIA_DRIVER_CAPABILITIES=all -v /etc/localtime:/etc/localtime:ro -v /root/autoware_map/sample-map-planning/:/autoware_map:ro ghcr.io/autowarefoundation/autoware:universe ros2 launch autoware_launch autoware.launch.xml map_path:=/autoware_map vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit
> 
> [INFO] [launch]: All log files can be found below /root/.ros/log/2025-02-28-11-48-47-446974-docker-desktop-1 [INFO] [launch]: Default logging verbosity is set to INFO [WARNING] [launch_ros.actions.node]: Parameter file path is not a file: /opt/ros/humble/share/ublox_gps/c94_f9p_rover.yaml [ERROR] [launch]: Caught exception in launch (see debug for traceback): "package 'autoware_probabilistic_occupancy_grid_map' not found, searching: ['/opt/autoware', '/opt/ros/humble']" [WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the '/sensing/lidar/right/pointcloud_preprocessor/pointcloud_container/_container/load_node' service, due to shutdown. [WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the '/map/map_container/_container/load_node' service, due to shutdown. [WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the '/system/system_monitor/system_monitor/system_monitor_container/_container/load_node' service response, due to shutdown. [WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the 'pointcloud_container/_container/load_node' service response, due to shutdown. [WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the '/pointcloud_container/_container/load_node' service response, due to shutdown. [WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the '/system/mrm_comfortable_stop_operator/mrm_comfortable_stop_operator_container/_container/load_node' service response, due to shutdown. [WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the 'pointcloud_container/_container/load_node' service response, due to shutdown. [WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the '/sensing/lidar/top/pointcloud_preprocessor/pointcloud_container/_container/load_node' service response, due to shutdown. [WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the '/pointcloud_container/_container/load_node' service response, due to shutdown. [WARNING] [launch_ros.actions.load_composable_ [... cut ...]
> Now, I saw your comment where you mentioned that after those commits, you could resolve that error. But those commits are inside the autoware.universe repository right? But the documentation asks us to clone this repository https://github.com/autowarefoundation/autoware.git
> 
> So how were you able to resolve the errors? I am on the latest pull of the repository mentioned above and in the main branch. Can you please elaborate on steps to resolve the error?

I think the latest build causes that issue,
So, now I'm testing with 0.39.1 version which is quite stable than newer version.

YuqiHuai, 2025-02-28:
I ended up using an older version too. A very old version.

ishit16, 2025-02-28:
> I think the latest build causes that issue, So, now I'm testing with 0.39.1 version which is quite stable than newer version.

Is it working for you? I am still getting the same error.

Just to confirm, you clone this repository https://github.com/autowarefoundation/autoware.git, and here you checkout 0.39.1 and follow the documentation steps, right? I did that and got the same error, I think the problem is with the image that it pulls when we run this command for the first time

./docker/run.sh --map-path path_to_map_data

ishit16, 2025-02-28:
> I ended up using an older version too. A very old version.

Which version are you using?

YuqiHuai, 2025-02-28:
I came from running Baidu Apollo and I tried to make running Autoware as similar as Apollo. So I ended up using this specific prebuilt image ghcr.io/autowarefoundation/autoware:20240903-devel-cuda-amd64

https://github.com/YuqiHuai/better_autoware_launch/blob/e9cc443e5e20cef003063d3caf15a57a96ae02fd/docker/build/container.dockerfile#L1

khanhnamdn91, 2025-03-03:
I run with Docker in main branch and got the same error. Does anyone know how to fix it?

ishit16, 2025-03-03:
I think there is a problem with their image, I tried to change the image to 0.39.1 in the ./docker/run.sh file but now I got the error because some other package is missing.

YuqiHuai, 2025-03-03:
Have you all considered trying this release? https://github.com/autowarefoundation/autoware/releases/tag/2025.02
If I remember correctly, I faced no issue on this release, but I am using the Cuda version.

moonseoppkim, 2025-03-03:
> I think there is a problem with their image, I tried to change the image to 0.39.1 in the ./docker/run.sh file but now I got the error because some other package is missing.

ghcr.io/autowarefoundation/autoware:universe-devel-0.39.1-amd64

I'm using this version, hopes this can help you.

khanhnamdn91, 2025-03-03:
> Have you all considered trying this release? https://github.com/autowarefoundation/autoware/releases/tag/2025.02 If I remember correctly, I faced no issue on this release, but I am using the Cuda version.

do you run in Docker or build from source? I checkout this branch and run docker and run into the same eror

YuqiHuai, 2025-03-03:
I attempted docker. I don’t recall facing this specific issue. :(

khanhnamdn91, 2025-03-03:
> I attempted docker. I don’t recall facing this specific issue. :(

did you use this specific image: ghcr.io/autowarefoundation/autoware:universe-20250213? 
Because this is the image the owner of this thread ran and had the issue

YuqiHuai, 2025-03-03:
@khanhnamdn91 sorry. I replied with the wrong information. I faced another setup issue on main branch. Using 2025.02, I could setup but not run “run.sh”.

I ended up using this older image ghcr.io/autowarefoundation/autoware:20240903-devel-cuda-amd64 and had no issues.

khanhnamdn91, 2025-03-03:
> [@khanhnamdn91](https://github.com/khanhnamdn91) sorry. I replied with the wrong information. I faced another setup issue on main branch. Using 2025.02, I could setup but not run “run.sh”.
>
> I ended [...]

thanks for your input. I will try to use the image you used there

ishit16, 2025-03-03:
@moonseoppkim  so you clone the repo and in run.sh you change the image name from what's mentioned there to ghcr.io/autowarefoundation/autoware:universe-devel-0.39.1-amd64 and it runs fine?

ishit16, 2025-03-03:
@moonseoppkim @YuqiHuai Can you all help me to understand where I am going wrong? First I clone this repo git clone https://github.com/autowarefoundation/autoware.git

and then I run this command inside the repo

./setup-dev-env.sh -y --no-nvidia docker

and it runs fine and installs some dependencies and packages

then I change the run.sh to use an older image as mentioned by both of you

[image]

I tried to use the image ghcr.io/autowarefoundation/autoware:universe-devel-0.39.1-amd64

and then I run this command 

./docker/run.sh --map-path ~/autoware_map/sample-map-planning/ --headless --no-nvidia

with the path to sample map.

and I get this error

Caught exception in launch (see debug for traceback): [Errno 2] No such file or directory: '/root/autoware_data/lidar_centerpoint/centerpoint_tiny_ml_package.param.yaml'

ishit16, 2025-03-04:
> thanks for your input. I will try to use the image you used there

Hey! Did you try and did it work?

YuqiHuai, 2025-03-04:
@ishit16 I faced issues when running autoware according to the latest documentation. So all I have been doing is just using ghcr.io/autowarefoundation/autoware:20240903-devel-cuda-amd64 and running
```
rocker --nvidia --privileged --x11 --user \
    --volume $GREAT_GRAND_PARENT/data/autoware_map:$HOME/autoware_map \
    --volume $GREAT_GRAND_PARENT:$HOME/better_autoware_launch \
    --name $CONTAINER_NAME \
    --mode non-interactive \
    --detach \
    -- $IMAGE_NAME sleep infinity
```
from my repo https://github.com/YuqiHuai/better_autoware_launch/blob/main/docker/scripts/dev_start.sh

My goal was to only get Autoware + Plan Sim up and running, and I have NVIDIA GPU on my machine. If that is your goal, you can give my scripts a try.

khanhnamdn91, 2025-03-05:
> Hey! Did you try and did it work?

I run with image ghcr.io/autowarefoundation/autoware:universe-devel-0.39.1-amd64 (for dev) and ghcr.io/autowarefoundation/autoware:universe-0.39.1-amd64 (for runtime), no cuda option. It works

jmlee0130, 2025-03-06:
> [@moonseoppkim](https://github.com/moonseoppkim) [@YuqiHuai](https://github.com/YuqiHuai) Can you all help me to understand where I am going wrong? First I clone this repo git clone [...]

I have the same problem, too. I run ./docker/run.sh  --no-nvidia --map-path ~/autoware_map/sample-map-planning
with docker image ghcr.io/autowarefoundation/autoware:universe-0.39.1-amd64, then I got the error [Errno 2] No such file or directory: '/root/autoware_data/lidar_centerpoint/centerpoint_tiny_ml_package.param.yaml'.
Why are Autoware's Docker images not properly usable? It seems like they need better maintenance."

moonseoppkim, 2025-03-06:
You should download autoware_data first, refering this guide in order to download this folder manually.
https://github.com/autowarefoundation/autoware/tree/main/ansible/roles/artifacts

* FYI) autoware_data folder contains some ML models for autoware simulations.

------------------------------------------------------------------------------------------------------------------------------
cd ~/autoware # The root directory of the cloned repository
ansible-galaxy collection install -f -r "ansible-galaxy-requirements.yaml"
ansible-playbook autoware.dev_env.download_artifacts -e "data_dir=$HOME/autoware_data" --ask-become-pass

YuqiHuai, 2025-03-06:
@jmlee0130 I created [a setup script](https://github.com/YuqiHuai/better_autoware_launch/blob/main/docker/scripts/setup.sh) to install autoware_map and autoware data, and modified Autoware's [run.sh](https://github.com/YuqiHuai/better_autoware_launch/blob/main/docker/scripts/run.sh) to use the 20250207 version. I tested this on my machine, both nvidia and no-nvidia works. Hope this helps!

jmlee0130, 2025-03-07:
> You should download autoware_data first, refering this guide in order to download this folder manually. https://github.com/autowarefoundation/autoware/tree/main/ansible/roles/artifacts
>
>     * FYI) [...]

Oh I've just missed it. Thanks for replying!

jmlee0130, 2025-03-07:
> [@jmlee0130](https://github.com/jmlee0130) I created [a setup script](https://github.com/YuqiHuai/better_autoware_launch/blob/main/docker/scripts/setup.sh) to install autoware_map and autoware data, [...]

Thanks for replying! I just cloned your github repository, and ran 
bash docker/scripts/setup.sh && bash docker/scripts/run.sh --no-nvidia.

Inside the container, have you tested planning simulation like "ros2 launch autoware_launch planning_simulator.launch.xml map_path:=/autoware_map vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit" with no-nvidia?

I tested planning simulatuion as above, and it errors with 

"[ERROR] [1741312147.246153729] [planning.scenario_planning.parking.costmap_generator]: "map" passed to lookupTransform argument target_frame does not exist."

[image]

YuqiHuai, 2025-03-07:
@jmlee0130 Yes, I tested with 
```
ros2 launch autoware_launch planning_simulator.launch.xml \
    map_path:=/autoware_map/sample-map-planning \
    vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit
```

jmlee0130, 2025-03-07:
@YuqiHuai Thank you, I changed map path as you mentioned,

but now I am encountering an error:

"[component_container_mt-35] [ERROR] [1741313130.748668361] [planning.scenario_planning.parking.costmap_generator]: Could not find a connection between 'map' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees."

Do you happen to know anything about this error? It also appears when using the ghcr.io/autowarefoundation/autoware:universe-0.39.1-amd64 image.

jmlee0130, 2025-03-07:
@YuqiHuai Thank you for your response!

YuqiHuai, 2025-04-10:
@jmlee0130 ... a month later I am now running into your identical issue

> Could not find a connection between 'map' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees."

Now I have no idea what happened...

JongmyeongLee, 2025-04-11:
@YuqiHuai 
It solves by setting path in Rviz.
I think that issue occurs in all stable versions, and it removes by setting path in rviz.

jmlee0130, 2025-04-11:
@YuqiHuai 
It solves by setting path in Rviz.
I think that issue occurs in all stable versions, and it removes by setting path in rviz.

thiagodeborba, 2025-05-13:
> [@moonseoppkim](https://github.com/moonseoppkim) [@YuqiHuai](https://github.com/YuqiHuai) Can you all help me to understand where I am going wrong? First I clone this repo git clone [...]

Could you solve this problem?