autowarefoundation/autoware #5785: Unable to run `setup-dev-env.sh docker`

YuqiHuai, 2025-02-14:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I'm convinced that this is not my fault but a bug.

### Description

I am running into `libcudnn8-dev : Depends: libcudnn8 (= 8.9.5.29-1+cuda12.2) but 8.9.7.29-1+cuda12.2 is to be installed` when running `setup-dev-env.sh` on the latest commit of Autoware. After trying to make some changes to `amd64.env`, I am then seeing other errors `libnvparsers-dev : Depends: libnvinfer-dev (= 8.6.1.6-1+cuda12.0) but 10.8.0.43-1+cuda12.8 is to be installed`

What might I be doing incorrectly?

### Expected behavior

`setup-dev-env.sh docker` sets up autoware for docker container usage.

### Actual behavior

```
fatal: [localhost]: FAILED! => {"cache_update_time": 1739514655, "cache_updated": true, "changed": false, "msg": "'/usr/bin/apt-get -y -o \"Dpkg::Options::=--force-confdef\" -o \"Dpkg::Options::=--force-confold\"       install 'libcudnn8=8.9.7.29-1+cuda12.2' 'libnvinfer10=10.8.0.43-1+cuda12.8' 'libnvinfer-plugin10=10.8.0.43-1+cuda12.8' 'libnvonnxparsers10=10.8.0.43-1+cuda12.8' --allow-downgrades --allow-change-held-packages' failed: E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.\n", "rc": 100, "stderr": "E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.\n", "stderr_lines": ["E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages."], "stdout": "Reading package lists...\nBuilding dependency tree...\nReading state information...\nSome packages could not be installed. This may mean that you have\nrequested an impossible situation or if you are using the unstable\ndistribution that some required packages have not yet been created\nor been moved out of Incoming.\nThe following information may help to resolve the situation:\n\nThe following packages have unmet dependencies:\n libcudnn8-dev : Depends: libcudnn8 (= 8.9.5.29-1+cuda12.2) but 8.9.7.29-1+cuda12.2 is to be installed\n", "stdout_lines": ["Reading package lists...", "Building dependency tree...", "Reading state information...", "Some packages could not be installed. This may mean that you have", "requested an impossible situation or if you are using the unstable", "distribution that some required packages have not yet been created", "or been moved out of Incoming.", "The following information may help to resolve the situation:", "", "The following packages have unmet dependencies:", " libcudnn8-dev : Depends: libcudnn8 (= 8.9.5.29-1+cuda12.2) but 8.9.7.29-1+cuda12.2 is to be installed"]}
```
(`libcudnn8-dev : Depends: libcudnn8 (= 8.9.5.29-1+cuda12.2) but 8.9.7.29-1+cuda12.2 is to be installed`)

```
fatal: [localhost]: FAILED! => {"cache_update_time": 1739515523, "cache_updated": true, "changed": false, "msg": "'/usr/bin/apt-get -y -o \"Dpkg::Options::=--force-confdef\" -o \"Dpkg::Options::=--force-confold\"       install 'libnvinfer-dev=10.8.0.43-1+cuda12.8' 'libnvinfer-plugin-dev=10.8.0.43-1+cuda12.8' 'libnvinfer-headers-dev=10.8.0.43-1+cuda12.8' 'libnvinfer-headers-plugin-dev=10.8.0.43-1+cuda12.8' 'libnvonnxparsers-dev=10.8.0.43-1+cuda12.8' --allow-downgrades --allow-change-held-packages' failed: E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.\n", "rc": 100, "stderr": "E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.\n", "stderr_lines": ["E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages."], "stdout": "Reading package lists...\nBuilding dependency tree...\nReading state information...\nSome packages could not be installed. This may mean that you have\nrequested an impossible situation or if you are using the unstable\ndistribution that some required packages have not yet been created\nor been moved out of Incoming.\nThe following information may help to resolve the situation:\n\nThe following packages have unmet dependencies:\n libnvparsers-dev : Depends: libnvinfer-dev (= 8.6.1.6-1+cuda12.0) but 10.8.0.43-1+cuda12.8 is to be installed\n", "stdout_lines": ["Reading package lists...", "Building dependency tree...", "Reading state information...", "Some packages could not be installed. This may mean that you have", "requested an impossible situation or if you are using the unstable", "distribution that some required packages have not yet been created", "or been moved out of Incoming.", "The following information may help to resolve the situation:", "", "The following packages have unmet dependencies:", " libnvparsers-dev : Depends: libnvinfer-dev (= 8.6.1.6-1+cuda12.0) but 10.8.0.43-1+cuda12.8 is to be installed"]}
```
(`libnvparsers-dev : Depends: libnvinfer-dev (= 8.6.1.6-1+cuda12.0) but 10.8.0.43-1+cuda12.8 is to be installed`)

### Steps to reproduce

1. Follow instructions on https://autowarefoundation.github.io/autoware-documentation/pr-631/installation/autoware/docker-installation/
2. ./setup-dev-env.sh -y docker

### Versions

- OS: Ubuntu 22.04
- autoware commmit 4374782
- NVIDIA-SMI 550.120
- Driver Version: 550.120
- CUDA Version: 12.4

maxime-clem, 2025-02-14:
**⚠ the following may break your environment so do it at your own risk ⚠** 

I had a similar issue recently and the solution is to completely remove all already installed nvidia packages with `sudo apt purge nvidia* libnv* cuda* libcudnn*`. After that you should be able to run the `setup-dev-env.sh` script to install the correct version of the Ubuntu packages.

YuqiHuai, 2025-02-14:
@maxime-clem Thanks for your response! This definitely sounds like something I should avoid. Switching to one of the older releases seems to have fixed the issue for me right now.