autowarefoundation/autoware #5804: Remove references to tamagawa_imu_driver

esteve, 2025-02-19:
> 3. Remove references of `tamagawa_imu_driver` from:

- https://github.com/autowarefoundation/awsim_labs_sensor_kit_launch/blob/6acfb178d9edac1f615daab71dbde5af50b71f57/awsim_labs_sensor_kit_launch/package.xml
- https://github.com/tier4/awsim_sensor_kit_launch/blob/c2e35f64ed0fb21bacf5d0cc78d40b4d3cc2119c/awsim_sensor_kit_launch/package.xml
-  https://github.com/autowarefoundation/single_lidar_sensor_kit_launch/blob/da146226cc681a0008f30ade5b02257cea2bab05/single_lidar_sensor_kit_launch/package.xml
-  https://github.com/autowarefoundation/sample_sensor_kit_launch/blob/10425283cc1dddd1f05a1478c96d48aad726bfd4/sample_sensor_kit_launch/package.xml
-  https://github.com/autowarefoundation/sample_sensor_kit_launch/blob/b798c920d35e0ade7616dd9795f6982c7400b5b1/sample_sensor_kit_launch/launch/imu.launch.xml (comment out)

esteve, 2025-02-19:
I can't fork `awsim_sensor_kit_launch` because it's already a fork of `awsim_labs_sensor_kit_launch` and GitHub doesn't let me. Once the changes in `awsim_sensor_kit_launch` are merged, I'll open a PR for the other repo.

xmfcx, 2025-02-20:
@esteve thanks for the PRs, all looked alright, merged them all.

Now only https://github.com/tier4/awsim_sensor_kit_launch is left 🏁

esteve, 2025-02-20:
@xmfcx thanks for reviewing all the PRs so quickly 🙏 I've submitted https://github.com/tier4/awsim_sensor_kit_launch/pull/32, once that is merged, I believe all the changes would be done.

youtalk, 2025-02-21:
@esteve Thank you for your contrinbution. The all sub issues were resolved. Let's close this issue.

esteve, 2025-02-24:
@youtalk references to `tamagawa_imu_driver` and `pacmod_interface` are removed from `autoware.repos` in https://github.com/autowarefoundation/autoware/pull/5801

I can bump up the versions, do you mean https://github.com/autowarefoundation/sample_sensor_kit_launch ?

esteve, 2025-03-10:
Closing, all related PRs have been merged.

closed 2025-03-10 by esteve