autowarefoundation/autoware #5826: Planning simulator failed with no CUDA

saidev27, 2025-02-26:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I'm convinced that this is not my fault but a bug.

### Description

Hi,

I found an issue while running the basic planning simulator (source installed). For some reason it failed with the below error.

1740438912.5269794 [ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'occupancy_grid_map_node' of type 'autoware::occupancy_grid_map::LaserscanBasedOccupancyGridMapNode' in container 'pointcloud_container': Component constructor threw an exception: cudaErrorNoDevice (100)@/home/saidev/autoware/install/autoware_cuda_utils/include/autoware/cuda_utils/cuda_unique_ptr.hpp#L43: no CUDA-capable device is detected
1740438912.5519762 [component_container_mt-1] [ERROR] [1740438912.520161410] [pointcloud_container]: Component constructor threw an exception: cudaErrorNoDevice (100)@/home/saidev/autoware/install/autoware_cuda_utils/include/autoware/cuda_utils/cuda_unique_ptr.hpp#L43: no CUDA-capable device is detected

I have set the initial and Goal pose, but the path is not being created and when I checked the logs found this issue. Posted about it on Discord and was directed to create this issue.

### Expected behavior

The simple planning simulator is supposed to run without CUDA or a GPU so this is a problem with Autoware.

### Actual behavior

Planning simulator failing with no CUDA error

### Steps to reproduce

1. Once the source installation and build are successful. Start the planning simulator.
2. Set the Start and Goal pose
3. check for the planned map to be generated (expected)

### Possible causes

I am working on VMware workstation pro, and it cannot access the GPU on my windows system. This can be cause issues in future (Any suggestion here would be great on how to work it out). But for now my understanding is that planning simulator doesnt need CUDA or GPU

amadeuszsz, 2025-03-04:
@knzo25 
Isn't it caused by latest feature addressed in https://github.com/autowarefoundation/autoware.universe/pull/9542?
I would appreciate if you can take a look 🙏🏻 I think planning simulator used to work without CUDA.

knzo25, 2025-03-04:
@saidev27 
Sorry, I though I had already fixed that in 
https://github.com/autowarefoundation/autoware.universe/commit/4823597cd6c5f05725bb782dcadd8ce08e28df58
Can you tell me exactly where in the node it dies? If you are not using the latest branch, can you tell me which one are you using?

saidev27, 2025-03-06:
@knzo25 
This is the exact Error from the log.
`1741224732.8004868 [component_container_mt-1] [ERROR] [1741224732.791601279] [pointcloud_container]: Component constructor threw an exception: cudaErrorNoDevice (100)@/home/saidev/autoware/install/autoware_cuda_utils/include/autoware/cuda_utils/cuda_unique_ptr.hpp#L43: no CUDA-capable device is detected
1741224732.8103609 [ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'occupancy_grid_map_node' of type 'autoware::occupancy_grid_map::LaserscanBasedOccupancyGridMapNode' in container 'pointcloud_container': Component constructor threw an exception: cudaErrorNoDevice (100)@/home/saidev/autoware/install/autoware_cuda_utils/include/autoware/cuda_utils/cuda_unique_ptr.hpp#L43: no CUDA-capable device is detected`

Looks like the node is - occupancy_grid_map_node

I made a pull on 24th Feb 2025 from main and haven't pulled since then.

Here is the latest log from which I attached the error.
[launch.log](https://github.com/user-attachments/files/19099732/launch.log)

saidev27, 2025-03-06:
Also, I am running on Ubuntu 22.04 with VMWare Workstation Pro on Win 11. My Laptop has a T550 NVIDIA GPU, but VMWare doesn't allow any GPU Passthrough or DDA. Can you suggest me how to proceed on Win 11, as later I might need the GPU access??