autowarefoundation/autoware #5840: No space left on device for docker-build workflow

xmfcx, 2025-03-05:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I'm convinced that this is not my fault but a bug.

### Description

- https://github.com/autowarefoundation/autoware/actions/runs/13546559639/job/37859542292?pr=5801
- https://github.com/autowarefoundation/autoware/actions/runs/13556055859/job/38212669829?pr=5819

```
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        73G   21G   53G  28% /
```
The available space is plenty.

It fails at an early stage too.

### Expected behavior

-

### Actual behavior

-

### Steps to reproduce

-

### Possible causes

- Did we add something in a near past?
- Have the docker build cache been overgrown?

mitsudome-r, 2025-03-06:
The only thing that I can think of is CUDA upgrade, but I'm not sure why it started to fail at this timing.

youtalk, 2025-03-06:
I hope https://github.com/autowarefoundation/autoware/pull/5830 resolves this issue.

mitsudome-r, 2025-03-06:
I have investigated the size of build-cache with the following commands.

Before:
`skopeo inspect --raw docker://ghcr.io/autowarefoundation/autoware-buildcache@sha256:60073907bce4f4fe9e7ce14a1d4d4cb3920a8aefe8089c4396897f1ad190c4fa | jq '[.manifests[].size] | add'`
I get about 11.5 GB

Latest:
`skopeo inspect --raw docker://ghcr.io/autowarefoundation/autoware-buildcache:amd64-main | jq '[.manifests[].size] | add'`
I get about 4.2 GB

xmfcx, 2025-03-07:
So this was caused by an overgrowing cache.

I'm not sure how to prevent it from overgrowing.
But a simple solution would be to remove the cache periodically, maybe weekly to prevent it from going over a size limit.
Or a daily workflow could check its size and remove it once it goes over a size limit.
I think daily checking is a solution until we have other ways to keep its size limited.
And checking its size can be done with the command @mitsudome-r shared above.

youtalk, 2025-03-21:
"No space left" happened again...
https://github.com/autowarefoundation/autoware/actions/runs/13983476297?pr=5906

xmfcx, 2025-03-21:
We can reduce the cache size and make it per pull_request event to increase the frequency, after all it's very fast.

xmfcx, 2025-03-21:
https://github.com/autowarefoundation/autoware/actions/runs/13985670703

[image]

The cache is already small yet it happened :/

xmfcx, 2025-03-21:
We could remove the artifacts again but the reason we cannot build these docker images in a whole 53GB of empty space is due to trying to create many images in the same workflow. Each image contains some duplicated information. Maybe it could be split into multiple workflows to run across machines.

Even if we remove the artifacts, which will reduce 1.7GB, and maybe some from the cache as well, as long as we keep adding new images to be built, it might grow larger again.

Building and pushing the main devel and runtime image workflow could be separated from building the multicontainer images would solve this issue.

Or buying self hosted solutions for this too.

youtalk, 2025-03-21:
> Each image contains some duplicated information. Maybe it could be split into multiple workflows to run across machines.

There’s still the issue of consuming a large number of runners, but it’s worth trying. I’ll try modifying it to run `universe`’s multi-stage process on individual runners.

xmfcx, 2025-03-21:
https://github.com/autowarefoundation/autoware/tree/main/docker#multi-stage-dockerfile-structure

[image]

> There’s still the issue of consuming a large number of runners

It could be done in 2 workflows if we change the dependencies.

In first workflow we build multi-containers.
In the second workflow we build the monolithic images. And they don't need to refer to each other.

Monolithic build will use the cache and will be the required workflow.
The multi-container build is for release, can be slow, without cache, could run in parallel or after the monolithic build.

youtalk, 2025-03-22:
I'm really appreciate it if you create the PRs. 
My simple fix PRs don't work properly. 
- https://github.com/autowarefoundation/autoware/pull/5913
- https://github.com/autowarefoundation/autoware/pull/5909
-

youtalk, 2025-03-22:
First of all, I made revert PRs #5915 and #5916 to resolve the disk problem ASAP.

xmfcx, 2025-03-22:
> I'd really appreciate it if you create the PRs.

Alright, I will look into it this week, thanks for the opportunity 👍