autowarefoundation/autoware #5865: Issue With building

Hateley123, 2025-03-11:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I'm convinced that this is not my fault but a bug.

### Description

I followed the instructions on your website, but when I get to the colcon build stage and get 
                                                                
Starting >>> launch_testing                                                    
--- stderr: autoware_point_types                                               
CMake Error at /home/ros2_humble/install/ament_cmake_target_dependencies/share/ament_cmake_target_dependencies/cmake/ament_target_dependencies.cmake:77 (message):
  ament_target_dependencies() the passed package name
  'point_cloud_msg_wrapper' was not found before
Call Stack (most recent call first):
  CMakeLists.txt:20 (ament_target_dependencies)

which means point_cloud_msg_wrapper was never installed when i downloaded the package. Is this a seperate ros package I need to find and install?

### Expected behavior

It should work

### Actual behavior

It crashes with the above issue

### Steps to reproduce

1) Followed Instructions on website
2) colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release

crash

### Versions

- Ubuntu 22
- ROS 2
- Autoware Humble branch

### Possible causes

I dont know thats why I am here.

maxime-clem, 2025-03-11:
When setting up your workspace (https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/source-installation/#how-to-set-up-a-workspace), did you successfully run the `rosdep` command ?

Hateley123, 2025-03-11:
Yes it says updated cache in /root/.ros/rosdep/sources.cache
 at the end of doing rosdep update and then #All required rosdeps installed successfully
 after i do rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO

Hateley123, 2025-03-11:
I am also on the humble branch for the autoware repository

maxime-clem, 2025-03-12:
Do you have other build issues ? Or is `autoware_point_types` the only package failing to build ?

Hateley123, 2025-03-12:
Currently as when it runs it causes that issue and then crashes the entire build and aborts other packages.

maxime-clem, 2025-03-12:
Please try add the option `--continue-on-error`  to `colcon build` then and please report what other build issues are occurring.

Hateley123, 2025-03-12:
auto war still failed here.
CMake Error at /home/ros2_humble/install/ament_cmake_target_dependencies/share/ament_cmake_target_dependencies/cmake/ament_target_dependencies.cmake:77 (message):
  ament_target_dependencies() the passed package name
  'point_cloud_msg_wrapper' was not found before
Call Stack (most recent call first):
  CMakeLists.txt:20 (ament_target_dependencies)

Other errors
--- stderr: autoware_utils                                  
In file included from /home/ros2_humble/src/autoware/src/core/autoware_utils/autoware_utils/include/autoware_utils/ros/managed_transform_buffer.hpp:21,
                 from /home/ros2_humble/src/autoware/src/core/autoware_utils/autoware_utils/include/autoware_utils/autoware_utils.hpp:38,
                 from /home/ros2_humble/src/autoware/src/core/autoware_utils/autoware_utils/src/autoware_utils.cpp:15:
/home/ros2_humble/install/autoware_utils_pcl/include/autoware_utils_pcl/managed_transform_buffer.hpp:20:10: fatal error: pcl_ros/transforms.hpp: No such file or directory
   20 | #include <pcl_ros/transforms.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/autoware_utils.dir/build.make:76: CMakeFiles/autoware_utils.dir/src/autoware_utils.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:157: CMakeFiles/autoware_utils.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< autoware_utils [20.1s, exited with code 2]
Finished <<< autoware_fake_test_node [35.6s]                
--- stderr: pointcloud_to_laserscan                         
/home/ros2_humble/src/autoware/src/universe/external/pointcloud_to_laserscan/src/pointcloud_to_laserscan_node.cpp:52:10: fatal error: pcl_conversions/pcl_conversions.h: No such file or directory
   52 | #include "pcl_conversions/pcl_conversions.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

Error 2" 
Starting >>> dummy_robot_bringup
--- stderr: eagleye_geo_pose_fusion                          
/home/ros2_humble/src/autoware/src/universe/external/eagleye/eagleye_util/geo_pose_fusion/src/geo_pose_fusion.cpp:32:10: fatal error: geographic_msgs/msg/geo_pose_with_covariance_stamped.hpp: No such file or directory
   32 | #include <geographic_msgs/msg/geo_pose_with_covariance_stamped.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Error 3

Starting >>> autoware_radar_tracks_noise_filter
--- stderr: tamagawa_imu_driver                             
/home/ros2_humble/src/autoware/src/sensor_component/external/tamagawa_imu_driver/src/tag_can_driver.cpp:49:10: fatal error: can_msgs/msg/frame.hpp: No such file or directory
   49 | #include "can_msgs/msg/frame.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/tag_can_driver.dir/build.make:76: CMakeFiles/tag_can_driver.dir/src/tag_can_driver.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:165: CMakeFiles/tag_can_driver.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< tamagawa_imu_driver [24.2s, exited with code 2]
Starting >>> autoware_scenario_simulator_v2_adapter
Finished <<< examples_rclcpp_multithreaded_executor [26.3s]
Finished <<< test_cli [13.9s]
Starting >>> autoware_test_node
Starting >>> autoware_topic_relay_controller                
Finished <<< autoware_diagnostic_graph_utils [57.7s]        
--- stderr: autoware_bluetooth_monitor
In file included from /home/ros2_humble/src/autoware/src/universe/autoware.universe/system/autoware_bluetooth_monitor/src/bluetooth_monitor.cpp:15:
/home/ros2_humble/src/autoware/src/universe/autoware.universe/system/autoware_bluetooth_monitor/include/autoware/bluetooth_monitor/bluetooth_monitor.hpp:20:10: fatal error: diagnostic_updater/diagnostic_updater.hpp: No such file or directory
   20 | #include <diagnostic_updater/diagnostic_updater.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/autoware_bluetooth_monitor.dir/build.make:76: CMakeFiles/autoware_bluetooth_monitor.dir/src/bluetooth_monitor.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:141: CMakeFiles/autoware_bluetooth_monitor.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< autoware_bluetooth_monitor [10.1s, exited with code 2]
Starting >>> eagleye_fix2kml
Starting >>> examples_rclcpp_minimal_action_client          
--- stderr: autoware_radar_scan_to_pointcloud2              
[... 84 lines cut ...]
In file included from /home/ros2_humble/src/autoware/src/sensor_component/ros2_socketcan/ros2_socketcan/src/socket_can_receiver_node.cpp:17:
/home/ros2_humble/src/autoware/src/sensor_component/ros2_socketcan/ros2_socketcan/include/ros2_socketcan/socket_can_receiver_node.hpp:32:10: fatal error: can_msgs/msg/frame.hpp: No such file or directory
   32 | #include "can_msgs/msg/frame.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/socket_can_receiver_node.dir/build.make:76: CMakeFiles/socket_can_receiver_node.dir/src/socket_can_receiver_node.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:191: CMakeFiles/socket_can_receiver_node.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ros2_socketcan [9.26s, exited with code 2]
Starting >>> ros2doctor
[... cut ...]  eagleye_can_velocity_converter eagleye_coordinate eagleye_geo_pose_fusion examples_rclpy_minimal_service nebula_decoders pointcloud_to_laserscan rclpy ros2_socketcan rosbag2_py tamagawa_imu_driver tier4_api_msgs tier4_map_msgs yabloc_monitor
  231 packages not processed

I am currently operating off of ubuntu 22.04 ros 2 humble and I am followin the directions on https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/source-installation/#how-to-set-up-a-workspace

maxime-clem, 2025-03-12:
It looks like you have many problems in your environment. Did you initially run the `setup-dev-env.sh` script without any problem ?

Hateley123, 2025-03-12:
Yes there were no issues up to the colcon build part of the website

Hateley123, 2025-03-12:
i used docker pull ubuntu:22.04 to use an official ubuntu 22 docker image, installed ros2 on it according to (https://docs.ros.org/en/humble/Installation.html) and it all built well. I followed the instructions according to your site with no issues up until the colcon build

maxime-clem, 2025-03-12:
> i used docker pull ubuntu:22.04 to use an official ubuntu 22 docker image

If you use docker, please follow the instructions from https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/docker-installation/