autowarefoundation/autoware #5920: Documentation Out of Date due to PR #5916

tswie, 2025-03-23:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I'm convinced that this is not my fault but a bug.

### Description

What is the intended usage now for running the runtime image? PR #5916 seems to make this documentation out of date: https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/docker-installation/

The Runtime can no longer run without the `data-path` flag and errors out as a result. It is not clear to me what to supply to  `data-path` in order to run the runtime.

### Expected behavior

The Autoware Docker Runtime image would successfully launch by executing `./docker/run.sh --map-path {my/autoware/map/path}`. 

### Actual behavior

I get the following message:

Note: The --map-path and --data-path option is mandatory for the universe(runtime image). For development environment with shell access, use --devel option.

### Steps to reproduce

1. Attempt to run the Autoware Runtime image by executing `./docker/run.sh --map-path {my/map/path}` as indicated by the documentation here: https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/docker-installation/

### Versions

- OS: 22.04
- Autoware: Main, last commit: 0091d4153a56577fa553c9bad954f3aa8603a670

### Possible causes

https://github.com/autowarefoundation/autoware/pull/5916

joeyjyyang, 2025-04-02:
Wondering the same. This is also affecting https://autowarefoundation.github.io/autoware-documentation/main/tutorials/components_evaluation/localization_evaluation/urban-environment-evaluation/#launch-autoware and https://autowarefoundation.github.io/autoware-documentation/main/tutorials/components_evaluation/localization_evaluation/urban-environment-evaluation/#launch-autoware

amadeuszsz, 2025-04-09:
Hi @tswie & @joeyjyyang 
There was a sequence of PR reverts [[1](https://github.com/autowarefoundation/autoware/pull/5916)][[2](https://github.com/autowarefoundation/autoware/pull/5915)], where the last one didn't aligned across all Autoware repositories [[3](https://github.com/autowarefoundation/autoware-documentation/pull/652)]. Sorry for any inconvenience. 

@xmfcx 
Did we decided whether we want to store artifacts in runtime images? I see that there is ongoing work regarding space issues, so I don't know if we plan another reverts after CI fix. If we plan to keep it as it is, please review this missing revert https://github.com/autowarefoundation/autoware-documentation/pull/659

xmfcx, 2025-04-11:
Hello @amadeuszsz , is this issue resolved?