autowarefoundation/autoware #5974: error with ./setup-dev-env.sh（agnocast-kmod if not found in dkms for version v1.0.2）

phoenixGG, 2025-04-02:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I'm convinced that this is not my fault but a bug.

### Description

[image]
Hi, my current Ubuntu system is 22.04.05. When I use the command ./setup-dev-env.sh to install dependencies, an error is shown as in the image. Could you please help me with how to resolve this?

### Expected behavior

FIX

### Actual behavior

Fix

### Steps to reproduce

1

sykwer, 2025-04-03:
@phoenixGG 

Due to the specifications of PPA, we have discovered that older releases are being removed when a new version is published. This issue is expected to be resolved by merging [this pull request](https://github.com/autowarefoundation/autoware/pull/5977). We have also found that the current release method using PPA is incompatible with semantic versioning, so we will fix it promptly. We apologize for the inconvenience.

sykwer, 2025-04-03:
We've discovered that due to the specifications of PPA, releasing a new version of agnocast-kmod or agnocast-heaphook causes the previous version to become unavailable, with no workaround. As a result, we'll be changing our release policy. Apologies for the lack of prior investigation into the PPA behavior.

We will provide separate packages for each major version—for example: agnocast-kmod-v1, agnocast-kmod-v2, agnocast-heaphook-v1, agnocast-heaphook-v2, and so on.