autowarefoundation/autoware #6180: Planning simulation startup failed

263868161, 2025-05-27:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I'm convinced that this is not my fault but a bug.

### Description

I encountered an error when starting the planning simulation.

### Expected behavior

If the startup is successful, pipeline execution should proceed.but the startup failed.

### Actual behavior

bug information:[scenario_test_runner.py-1] [WARN] [1748329703.520345256] [simulation.scenario_test_runner]: openscenario_interpreter/get_state service unavailable
[scenario_test_runner.py-1] [WARN] [1748329704.523647665] [simulation.scenario_test_runner]: openscenario_interpreter/get_state service unavailable
[scenario_test_runner.py-1] [INFO] [1748329705.292674194] [simulation.scenario_test_runner]: connection established with preprocessor
[scenario_test_runner.py-1] Error: failed validating '1' with XsdPatternFacets(['[$][{][ A-Za-z0-9_\\+\\-\\*/%$\\(\\)\\.,]*[\\}]']):
[scenario_test_runner.py-1] 
[scenario_test_runner.py-1] Reason: value doesn't match any pattern of ['[$][{][ A-Za-z0-9_\\+\\-\\*/%$\\(\\)\\.,]*[\\}]']
[scenario_test_runner.py-1] 
[scenario_test_runner.py-1] Schema component:
[scenario_test_runner.py-1] 
[scenario_test_runner.py-1]   <xsd:pattern xmlns:xsd="http://www.w3.org/2001/XMLSchema" value="[$][{][ A-Za-z0-9_\+\-\*/%$\(\)\.,]*[\}]" />
[scenario_test_runner.py-1] 
[scenario_test_runner.py-1] Instance type: <class 'xml.etree.ElementTree.Element'>
[scenario_test_runner.py-1] 
[scenario_test_runner.py-1] Instance:
[scenario_test_runner.py-1] 
[scenario_test_runner.py-1]   <FileHeader author="Tatsuya Yamasaki" date="2022-03-04T18:06:53+09:00" description="Sample scenario (with Autoware)" revMajor="1" revMinor="0" />
[scenario_test_runner.py-1] 
[scenario_test_runner.py-1] Path: /OpenSCENARIO/FileHeader
[scenario_test_runner.py-1] 
[INFO] [scenario_test_runner.py-1]: process has finished cleanly [pid 940868]
[INFO] [launch]: process[scenario_test_runner.py-1] was required: shutting down launched system
[INFO] [visualization_node-5]: sending signal 'SIGINT' to process[visualization_node-5]
[INFO] [openscenario_preprocessor_node-4]: sending signal 'SIGINT' to process[openscenario_preprocessor_node-4]
[INFO] [openscenario_interpreter_node-3]: sending signal 'SIGINT' to process[openscenario_interpreter_node-3]
[INFO] [simple_sensor_simulator_node-2]: sending signal 'SIGINT' to process[simple_sensor_simulator_node-2]
[visualization_node-5] [INFO] [1748329706.396653586] [rclcpp]: signal_handler(signum=2)
[openscenario_preprocessor_node-4] [INFO] [1748329706.397146558] [rclcpp]: signal_handler(signum=2)
[openscenario_interpreter_node-3] [INFO] [1748329706.401956651] [rclcpp]: signal_handler(signum=2)
[simple_sensor_simulator_node-2] [INFO] [1748329706.410356872] [rclcpp]: signal_handler(signum=2)

### Steps to reproduce

command:ros2 launch scenario_test_runner scenario_test_runner.launch.py \
  architecture_type:=awf/universe/20240605 \
  record:=false \
  scenario:='$(find-pkg-share scenario_test_runner)/scenario/sample.yaml' \
  sensor_model:=sample_sensor_kit \
  vehicle_model:=sample_vehicle

### Versions

ubuntu:22.04.5

ros:humble

autoware:0.44.1

felixf4xu, 2025-06-12:
read the error log:

>[openscenario_interpreter_node-3] [ERROR] [launch]: Caught exception in launch (see debug for traceback): "package 'tier4_deprecated_api_adapter' not found

check if `tier4_deprecated_api_adapter` is used and exists

263868161, 2025-06-13:
> read the error log:
> 
> > [openscenario_interpreter_node-3] [ERROR] [launch]: Caught exception in launch (see debug for traceback): "package 'tier4_deprecated_api_adapter' not found
> 
> check if `tier4_deprecated_api_adapter` is used and exists

The package tier4_deprecated_api_adapter indeed does not exist.This is because the version of the tier4_ad_api_adaptor package in Autoware 0.44.1 is incorrect.
autoware.repos：
[image]

The correct version should be 0.43.0.

263868161, 2025-06-13:
I have solved the above problems, but a new problem has arisen.
The error message is as follows:
[openscenario_interpreter_node-3] [ERROR] [1749785308.467404389] [simulation.concealer_user]: The wait for Autoware launch process termination was interrupted by an unblocked, caught signal.
[openscenario_interpreter_node-3] [WARNING] [launch]: user interrupted with ctrl-c (SIGINT) again, ignoring...
[openscenario_interpreter_node-3] [INFO] [initial_pose_adaptor_node-54]: process has finished cleanly [pid 26381]
[scenario_test_runner.py-1] [WARN] [1749785308.472205038] [simulation.scenario_test_runner]: KeyboardInterrupt
[scenario_test_runner.py-1] Failed to publish log message to rosout: publisher's context is invalid, at ./src/rcl/publisher.c:389
[openscenario_interpreter_node-3] [INFO] [manual_status_node-50]: sending signal 'SIGINT' to process[manual_status_node-50]
[openscenario_interpreter_node-3] [INFO] [autoware_path_distance_calculator_node-49]: sending signal 'SIGINT' to process[autoware_path_distance_calculator_node-49]
[openscenario_interpreter_node-3] [INFO] [relay-48]: sending signal 'SIGINT' to process[relay-48]
[openscenario_interpreter_node-3] [INFO] [relay-47]: sending signal 'SIGINT' to process[relay-47]
[openscenario_interpreter_node-3] [INFO] [relay-46]: sending signal 'SIGINT' to process[relay-46]
[openscenario_interpreter_node-3] [ERROR] [rviz2-57]: process has died [pid 26399, exit code -2, cmd '/opt/ros/humble/lib/rviz2/rviz2 -d /home/legion/workspace/autoware/autoware-0.44.1/install/traffic_simulator/share/traffic_simulator/config/scenario_simulator_v2.rviz -s /home/legion/workspace/autoware/autoware-0.44.1/install/autoware_launch/share/autoware_launch/rviz/image/autoware.png --ros-args -r __node:=rviz2 -p use_sim_time:=False -p wheel_radius:=0.383 -p wheel_width:=0.235 -p wheel_base:=2.79 -p wheel_tread:=1.64 -p front_overhang:=1.0 -p rear_overhang:=1.1 -p left_overhang:=0.128 -p right_overhang:=0.128 -p vehicle_height:=2.5 -p max_steer_angle:=0.7'].
[openscenario_interpreter_node-3] [INFO] [component_container_mt-45]: sending signal 'SIGINT' to process[component_container_mt-45]
[openscenario_interpreter_node-3] [INFO] [component_container_mt-44]: sending signal 'SIGINT' to process[component_container_mt-44]
[openscenario_interpreter_node-3] [INFO] [component_container_mt-43]: sending signal 'SIGINT' to process[component_container_mt-43]
[openscenario_interpreter_node-3] [INFO] [manual_control_node-52]: process has finished cleanly [pid 26369]
[openscenario_interpreter_node-3] [INFO] [awapi_awiv_adapter-42]: sending signal 'SIGINT' to process[awapi_awiv_adapter-42]
[openscenario_interpreter_node-3] [INFO] [control_evaluator-41]: sending signal 'SIGINT' to process[control_evaluator-41]
[openscenario_interpreter_node-3] [INFO] [component_container_mt-40]: sending signal 'SIGINT' to process[component_container_mt-40]
[openscenario_interpreter_node-3] [INFO] [component_container_mt-39]: sending signal 'SIGINT' to process[component_container_mt-39]
[openscenario_interpreter_node-3] [INFO] [autoware_remaining_distance_time_calculator_node-38]: sending signal 'SIGINT' to process[autoware_remaining_distance_time_calculator_node-38]
[openscenario_interpreter_node-3] [INFO] [planning_evaluator-37]: sending signal 'SIGINT' to process[planning_evaluator-37]
[openscenario_interpreter_node-3] [INFO] [planning_validator_node-36]: sending signal 'SIGINT' to process[planning_validator_node-36]
[openscenario_interpreter_node-3] [INFO] [component_container_mt-35]: sending signal 'SIGINT' to process[component_container_mt-35]
[openscenario_interpreter_node-3] [INFO] [component_container_mt-34]: sending signal 'SIGINT' to process[component_container_mt-34]
[openscenario_interpreter_node-3] [INFO] [component_container_mt-33]: sending signal 'SIGINT' to process[component_container_mt-33]
[openscenario_interpreter_node-3] [INFO] [component_container-32]: sending signal 'SIGINT' to process[component_container-32]
[openscenario_interpreter_node-3] [INFO] [external_velocity_limit_selector-31]: sending signal 'SIGINT' to process[external_velocity_limit_selector-31]
[openscenario_interpreter_node-3] [INFO] [autoware_scenario_selector_node-30]: sending signal 'SIGINT' to process[autoware_scenario_selector_node-30]
[openscenario_interpreter_node-3] [INFO] [goal_pose_visualizer-29]: sending signal 'SIGINT' to process[goal_pose_visualizer-29]
[openscenario_interpreter_node-3] [INFO] [component_container_mt-28]: sending signal 'SIGINT' to process[component_container_mt-28]
[openscenario_interpreter_node-3] [INFO] [autoware_map_projection_loader_node-27]: sending signal 'SIGINT' to process[autoware_map_projection_loader_node-27]
[openscenario_interpreter_node-3] [INFO] [map_hash_generator-26]: sending signal 'SIGINT' to process[map_hash_generator-26]
[openscenario_interpreter_node-3] [INFO] [component_container_mt-25]: sending signal 'SIGINT' to process[component_container_mt-25]
[openscenario_interpreter_node-3] [INFO] [autoware_dummy_diag_publisher_node-24]: sending signal 'SIGINT' to process[autoware_dummy_diag_publisher_node-24]
[openscenario_interpreter_node-3] [ERROR] [map_hash_generator-26]: process has died [pid 26152, exit code -2, cmd '/home/legion/workspace/autoware/autoware-0.44.1/install/autoware_map_loader/lib/autoware_map_loader/map_hash_generator --ros-args -r __node:=map_hash_generator -r __ns:=/map -p use_sim_time:=False -p wheel_radius:=0.383 -p wheel_width:=0.235 -p wheel_base:=2.79 -p whe [... cut ...]
[openscenario_interpreter_node-3] [ERROR] [autoware_duplicated_node_checker_node-3]: process[autoware_duplicated_node_checker_node-3] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[openscenario_interpreter_node-3] [ERROR] [robot_state_publisher-2]: process[robot_state_publisher-2] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[openscenario_interpreter_node-3] [ERROR] [component_container_mt-1]: process[component_container_mt-1] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'

felixf4xu, 2025-06-16:
Your description is misleading, I guess you are running scenario simulation, not planning simulation, could you post the command that you start your simulation?

If you are running scenario simulation, a higher CPU capacity is required, could you check the current CPU load when it fails? There are several parameters that you can change for different CPU capacity, if you post the command & parameters.

felixf4xu, 2025-06-16:
the command supports several parameters, if your PC is not powerful enough, try to increase `initialize_duration` and descrease `global_frame_rate`

The parameter format is like:

>ros2 launch scenario_test_runner scenario_test_runner.launch.py architecture_type:=awf/universe record:=false scenario:=/autoware/test-scenario/yield_maneuver_demo.yaml sensor_model:=sample_sensor_kit vehicle_model:=sample_vehicle map_path:=/autoware/test-scenario/map launch_autoware:=False **initialize_duration:=60 global_frame_rate:=12**

263868161, 2025-06-18:
> the command supports several parameters, if your PC is not powerful enough, try to increase `initialize_duration` and descrease `global_frame_rate`
>
> The parameter format is like:

I execute based on your command:
ros2 launch scenario_test_runner scenario_test_runner.launch.py architecture_type:=awf/universe/20240605 record:=false scenario:='$(find-pkg-share scenario_test_runner)/scenario/sample.yaml' sensor_model:=sample_sensor_kit vehicle_model:=sample_vehicle  launch_autoware:=False initialize_duration:=60 global_frame_rate:=12

The result is as follows:
[scenario_test_runner.py-1] [INFO] [1750207938.761043330] [simulation.scenario_test_runner]: connection established with preprocessor
[openscenario_preprocessor_node-4] /tmp/converted_scenario/vehicle/sample_vehicle.xosc
[scenario_test_runner.py-1] [INFO] [1750207942.253165631] [simulation.scenario_test_runner]: Result of /simulation/openscenario_preprocessor/load: success
[scenario_test_runner.py-1] [INFO] [1750207942.255302971] [simulation.scenario_test_runner]: Result of /simulation/openscenario_preprocessor/check: True
[scenario_test_runner.py-1] [INFO] [1750207942.257328210] [simulation.scenario_test_runner]: derived : /tmp/scenario_test_runner/sample/sample_0.xosc
[scenario_test_runner.py-1] [INFO] [1750207942.259107913] [simulation.scenario_test_runner]: Result of /simulation/openscenario_preprocessor/check: False
[scenario_test_runner.py-1] [INFO] [1750207942.260034897] [simulation.scenario_test_runner]: finish derivation
[scenario_test_runner.py-1] [INFO] [1750207942.260933909] [simulation.scenario_test_runner]: /tmp/scenario_test_runner/sample/sample_0.xosc
[scenario_test_runner.py-1] [INFO] [1750207942.261838221] [simulation.scenario_test_runner]: Run sample_0.xosc (1 of 1)
[openscenario_interpreter_node-3] /tmp/converted_scenario/vehicle/sample_vehicle.xosc
[openscenario_interpreter_node-3] [WARN] [1750207946.105302246] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[openscenario_interpreter_node-3] [WARN] [1750207946.136884310] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[simple_sensor_simulator_node-2] [WARN] [1750207946.188756803] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[simple_sensor_simulator_node-2] [WARN] [1750207946.206732372] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[simple_sensor_simulator_node-2] [WARN] [1750207946.238255686] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[scenario_test_runner.py-1] [ERROR] [1750208126.677529188] [simulation.scenario_test_runner]: The simulation has timed out. Forcibly inactivate.
[openscenario_interpreter_node-3] main of openscenario_interpreter_node is probably unresponsive.
[openscenario_interpreter_node-3] main of openscenario_interpreter_node is probably unresponsive.
[openscenario_interpreter_node-3] main of openscenario_interpreter_node is probably unresponsive.
[openscenario_interpreter_node-3] main of openscenario_interpreter_node is probably unresponsive.
[openscenario_interpreter_node-3] main of openscenario_interpreter_node is probably unresponsive.
[openscenario_interpreter_node-3] main of openscenario_interpreter_node is probably unresponsive.
[openscenario_interpreter_node-3] main of openscenario_interpreter_node is probably unresponsive.
[openscenario_interpreter_node-3] main of openscenario_interpreter_node is probably unresponsive.
[openscenario_interpreter_node-3] main of openscenario_interpreter_node is probably unresponsive.
[openscenario_interpreter_node-3] main of openscenario_interpreter_node is probably unresponsive.