autowarefoundation/autoware #6263: Autoware Development Environment Setup Failed - Missing Linux Headers for Tegra Kernel

Ablazesphere, 2025-06-26:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I'm convinced that this is not my fault but a bug.

### Description

-Problem Summary
Autoware development environment setup fails during the installation of linux headers for the running kernel on NVIDIA Jetson device.

-System Information
Platform: NVIDIA Jetson AGX Orin 64GB Developer Kit (Ubuntu Jammy)
Kernel Version: 5.15.136-tegra
ROS Distribution: Humble
Autoware Repository: Universe
Setup Command: sudo ./setup-dev-env.sh --no-nvidia --no-cuda-drivers

-What I Was Trying to Do
Setting up the Autoware development environment using the official setup script with NVIDIA drivers disabled (since running on Jetson hardware).

-Error Details
The setup process fails at the following task:
TASK [autoware.dev_env.agnocast : Install linux headers for the running kernel]
fatal: [localhost]: FAILED! => {"changed": false, "msg": "No package matching 'linux-headers-5.15.136-tegra' is available"}

### Expected behavior

-

### Actual behavior

TASK [autoware.dev_env.agnocast : Install linux headers for the running kernel]
fatal: [localhost]: FAILED! => {"changed": false, "msg": "No package matching 'linux-headers-5.15.136-tegra' is available"}

### Steps to reproduce

sudo ./setup-dev-env.sh --no-nvidia --no-cuda-drivers

### Versions

Platform: NVIDIA Jetson AGX Orin 64GB Developer Kit (Ubuntu Jammy)
Kernel Version: 5.15.136-tegra
ROS Distribution: Humble
Autoware Repository: Universe

### Additional context

[logs.txt](https://github.com/user-attachments/files/20924113/logs.txt)

G7-3208, 2025-07-01:
I encountered the same issue.

Ablazesphere, 2025-07-01:
[image]

Remove this line and then run the ansible script again `./setup-dev-env.sh --no-nvidia --no-cuda-drivers`

Ablazesphere, 2025-07-01:
> thank you for your reply .is it in the file”setup-dev-env.sh“?

~/autoware/ansible/playbooks/universe.yaml

G7-3208, 2025-07-01:
Thank you again for your reply. I have a few more questions to ask you. Firstly, may I ask why the command "./setup dev env. sh -- no Nvidia -- no CUDA driver" needs to include -- no Nvidia -- no CUDA driver. Then, what is Discord Community and I want to know how to join so that I can seek help when I encounter other issues.

G7-3208, 2025-07-02:
thank you so much @Ablazesphere ,I have now successfully run the command: /setup dev env. sh -- no Nvidia -- no CUDA driver， But I encountered the following error while running the next command：./docker/run.sh --devel

Digest: sha256:34fff46b29969686c08bb683920a3ac61f5fed9f4529700312d58556efe0a33b
Status: Downloaded newer image for ghcr.io/autowarefoundation/autoware:universe-devel-cuda
docker: Error response from daemon: failed to create task for container: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error running prestart hook #0: exit status 1, stdout: , stderr: Auto-detected mode as 'csv'
invoking the NVIDIA Container Runtime Hook directly (e.g. specifying the docker --gpus flag) is not supported. Please use the NVIDIA Container Runtime (e.g. specify the --runtime=nvidia flag) instead.: unknown

Ablazesphere, 2025-07-02:
> thank you so much [@Ablazesphere](https://github.com/Ablazesphere) ,I have now successfully run the command: /setup dev env. sh -- no Nvidia -- no CUDA driver， But I encountered the following error [...]

I'm following the 'Source Installation' option rather than using docker. So i dont have any solution for the error that you're facing. Maybe you can try asking somone in the discord server.

G7-3208, 2025-07-07:
I encountered three errors while building:--- stderr: autoware_tensorrt_plugins
CMake Warning at CMakeLists.txt:74 (find_package):
  By not providing "Findcumm.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "cumm", but
  CMake did not find one.

  Could not find a package configuration file provided by "cumm" with any of
  the following names:

    cummConfig.cmake
    cumm-config.cmake

  Add the installation prefix of "cumm" to CMAKE_PREFIX_PATH or set
  "cumm_DIR" to a directory containing one of the above files.  If "cumm"
  provides a separate development package or SDK, be sure it has been
  installed.

CMake Warning at CMakeLists.txt:75 (find_package):
  By not providing "Findspconv.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "spconv", but
  CMake did not find one.

  Could not find a package configuration file provided by "spconv" with any
  of the following names:

    spconvConfig.cmake
    spconv-config.cmake

  Add the installation prefix of "spconv" to CMAKE_PREFIX_PATH or set
  "spconv_DIR" to a directory containing one of the above files.  If "spconv"
  provides a separate development package or SDK, be sure it has been
  installed.

spconv is NOT available
CMake Error at CMakeLists.txt:128 (add_library):
  Target "autoware_tensorrt_plugins" links to target "CUDA::cudart" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:128 (add_library):
  Target "autoware_tensorrt_plugins" links to target "spconv::spconv" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Generate step failed.  Build files cannot be regenerated correctly.
---

--- stderr: yabloc_image_processing
In file included from /home/ll/autoware/src/universe/autoware_universe/localization/yabloc/yabloc_image_processing/src/graph_segment/graph_segment_core.cpp:15:
/home/ll/autoware/src/universe/autoware_universe/localization/yabloc/yabloc_image_processing/include/yabloc_image_processing/graph_segment/graph_segment.hpp:20:10: fatal error: opencv4/opencv2/ximgproc/segmentation.hpp: No such file or directory
   20 | #include <opencv4/opencv2/ximgproc/segmentation.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/yabloc_image_processing.dir/build.make:90: CMakeFiles/yabloc_image_processing.dir/src/graph_segment/graph_segment_core.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:149: CMakeFiles/yabloc_image_processing.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< yabloc_image_processing [18.5s, exited with code 2]

Starting >>> autoware_mpc_lateral_controller
--- stderr: autoware_elevation_map_loader
/usr/bin/ld: cannot find -lopencv_alphamat: No such file or directory
/usr/bin/ld: cannot find -lopencv_aruco: No such file or directory
/usr/bin/ld: cannot find -lopencv_barcode: No such file or directory
/usr/bin/ld: cannot find -lopencv_bgsegm: No such file or directory
/usr/bin/ld: cannot find -lopencv_bioinspired: No such file or directory
/usr/bin/ld: cannot find -lopencv_ccalib: No such file or directory
/usr/bin/ld: cannot find -lopencv_datasets: No such file or directory
[... 14 lines cut ...]
/usr/bin/ld: cannot find -lopencv_plot: No such file or directory
/usr/bin/ld: cannot find -lopencv_quality: No such file or directory
/usr/bin/ld: cannot find -lopencv_rapid: No such file or directory
/usr/bin/ld: cannot find -lopencv_reg: No such file or directory
/usr/bin/ld: cannot find -lopencv_rgbd: No such file or directory
/usr/bin/ld: cannot find -lopencv_saliency: No such file or directory
/usr/bin/ld: cannot find -lopencv_shape: No such file or directory
/usr/bin/ld: cannot find -lopencv_stereo: No such file or directory
/usr/bin/ld: cannot find -lopencv_structured_light: No such file or directory
/usr/bin/ld: cannot find -lopencv_superres: No such file or directory
/usr/bin/ld: cannot find -lopencv_surface_matching: No such file or directory
/usr/bin/ld: cannot find -lopencv_text: No such file or directory
/usr/bin/ld: cannot find -lopencv_tracking: No such file or directory
/usr/bin/ld: cannot find -lopencv_videostab: No such file or directory
/usr/bin/ld: cannot find -lopencv_viz: No such file or directory
/usr/bin/ld: cannot find -lopencv_wechat_qrcode: No such file or directory
/usr/bin/ld: cannot find -lopencv_ximgproc: No such file or directory
/usr/bin/ld: cannot find -lopencv_xobjdetect: No such file or directory
/usr/bin/ld: cannot find -lopencv_xphoto: No such file or directory
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/autoware_elevation_map_loader.dir/build.make:795: libautoware_elevation_map_loader.so] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:139: CMakeFiles/autoware_elevation_map_loader.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< autoware_elevation_map_loader [4.41s, exited with code 2]

G7-3208, 2025-07-18:
yes, you can use the command "dpkg - s libOpenCV dev | grep Version" to check the OpenCV version. For me, my mistake was that there were two software sources, and the system automatically selected version 4.8 of OpenCV, which seems incomplete. Therefore, choosing version 4.5 of OpenCV can solve the problem.

Ablazesphere, 2025-07-18:
> yes, you can use the command "dpkg - s libOpenCV dev | grep Version" to check the OpenCV version. For me, my mistake was that there were two software sources, and the system automatically selected [...]

How do i choose 4.5 version of openCV. Could you give me some instructions please?

G7-3208, 2025-07-18:
of course.firstly，use the command ”apt-cache policy libopencv-dev“ to check which versions are available in your software sources .Then,block other versions and only select  version 4.5.Do you have any social media accounts? I can add you for easier communication

Ablazesphere, 2025-07-18:
https://github.com/Ablazesphere/autoware_AGX/blob/main/README.md

To save others from the same frustration, I've documented the process in a guide for building and installing Autoware 0.44.2 on the AGX Orin. I plan to update it with instructions for version 0.45 once I find a working solution.