autowarefoundation/autoware #6292: Enable generator-selector planning framework

xmfcx, 2025-07-09:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I've agreed with the maintainers that I can plan this task.

### Description

-

### Purpose

-

### Possible approaches

-

### Definition of done

-

yukkysaito, 2025-08-08:
>🔴 Phase 2 | 📅 2025-08-11 | Update Controller Input Topic
https://github.com/autowarefoundation/autoware_launch/pull/1594
https://github.com/autowarefoundation/autoware_core/pull/602
https://github.com/autowarefoundation/autoware_universe/pull/11135
https://github.com/tier4/tier4_ad_api_adaptor/pull/169

I have merged the above PR and confirmed that it works as expected.

[image]

[image]

mitsudome-r, 2025-08-08:
@yukkysaito Could you also create a pr to autoware_tools repository as well?

xmfcx, 2025-08-09:
@yukkysaito I've checked for the usages of `/planning/scenario_planning/trajectory` and they all seem to be covered by your PRs. I've found one usage as a part of an rviz config in the scenario_simulator_v2 and created a PR for that:
- https://github.com/tier4/scenario_simulator_v2/pull/1664